pypose/pypose #109: error when install pypose: /torch/include/torch/csrc/api/include/torch/nn/cloneable.h:57:59: error

StiphyJay, 2022-08-07:
When I follow your instruction to install pypyose, I meet the error:

[with Derived = torch::nn::CrossMapLRN2dImpl]’: /tmp/tmpxft_0000d2f7_00000000-6_basis_cuda.cudafe1.stub.c:51:27: required from here /disk1/zsxm/miniconda3/envs/pytorch/lib/python3.7/site-packages/torch/include/torch/csrc/api/include/torch/nn/cloneable.h:57:59: error: invalid static_cast from type ‘const torch::OrderedDict<std::basic_string<char>, at::Tensor>’ to type ‘torch::OrderedDict<std::basic_string<char>, at::Tensor>&’ /disk1/zsxm/miniconda3/envs/pytorch/lib/python3.7/site-packages/torch/include/torch/csrc/api/include/torch/nn/cloneable.h:69:61: error: invalid static_cast from type ‘const torch::OrderedDict<std::basic_string<char>, std::shared_ptr<torch::nn::Module> >’ to type ‘torch::OrderedDict<std::basic_string<char>, std::shared_ptr<torch::nn::Module> >&’ /disk1/zsxm/miniconda3/envs/pytorch/lib/python3.7/site-packages/torch/include/torch/csrc/api/include/torch/nn/cloneable.h: In instantiation of ‘std::shared_ptr<torch::nn::Module> torch::nn::Cloneable<Derived>::clone(const c10::optional<c10::Device>&) const [with Derived = torch::nn::EmbeddingBagImpl]’: /tmp/tmpxft_0000d2f7_00000000-6_basis_cuda.cudafe1.stub.c:51:27: required from here /disk1/zsxm/miniconda3/envs/pytorch/lib/python3.7/site-packages/torch/include/torch/csrc/api/include/torch/nn/cloneable.h:57:59: error: invalid static_cast from type ‘const torch::OrderedDict<std::basic_string<char>, at::Tensor>’ to type ‘torch::OrderedDict<std::basic_string<char>, at::Tensor>&’ /disk1/zsxm/miniconda3/envs/pytorch/lib/python3.7/site-packages/torch/include/torch/csrc/api/include/torch/nn/cloneable.h:69:61: error: invalid static_cast from type ‘const torch::OrderedDict<std::basic_string<char>, std::shared_ptr<torch::nn::Module> >’ to type ‘torch::OrderedDict<std::basic_string<char>, std::shared_ptr<torch::nn::Module> >&’ /disk1/zsxm/miniconda3/envs/pytorch/lib/python3.7/site-packages/torch/include/torch/csrc/api/include/torch/nn/cloneable.h: In instantiation of ‘std::shared_ptr<torch::nn::Module> torch::nn::Cloneable<Derived>::clone(const c10::optional<c10::Device>&) const [with Derived = torch::nn::EmbeddingImpl]’: /tmp/tmpxft_0000d2f7_00000000-6_basis_cuda.cudafe1.stub.c:51:27: required from here /disk1/zsxm/miniconda3/envs/pytorch/lib/python3.7/site-packages/torch/include/torch/csrc/api/include/torch/nn/cloneable.h:57:59: error: invalid static_cast from type ‘const torch::OrderedDict<std::basic_string<char>, at::Tensor>’ to type ‘torch::OrderedDict<std::basic_string<char>, at::Tensor>&’ /disk1/zsxm/miniconda3/envs/pytorch/lib/python3.7/site-packages/torch/include/torch/csrc/api/include/torch/nn/cloneable.h:69:61: error: invalid static_cast from type ‘const torch::OrderedDict<std::basic_string<char>, std::shared_ptr<torch::nn::Module> >’ to type ‘torch::OrderedDict<std::basic_string<char>, std::shared_ptr<torch::nn::Module> >&’ /disk1/zsxm/miniconda3/envs/pytorch/lib/python3.7/site-packages/torch/include/torch/csrc/api/include/torch/nn/cloneable.h: In instantiation of ‘std::shared_ptr<torch::nn::Module> torch::nn::Cloneable<Derived>::clone(const c10::optional<c10::Device>&) const [with Derived = torch::nn::ParameterDictImpl]’: /tmp/tmpxft_0000d2f7_00000000-6_basis_cuda.cudafe1.stub.c:51:27: required from here /disk1/zsxm/miniconda3/envs/pytorch/lib/python3.7/site-packages/torch/include/torch/csrc/api/include/torch/nn/cloneable.h:57:59: error: invalid static_cast from type ‘const torch::OrderedDict<std::basic_string<char>, at::Tensor>’ to type ‘torch::OrderedDict<std::basic_string<char>, at::Tensor>&’ /disk1/zsxm/miniconda3/envs/pytorch/lib/python3.7/site-packages/torch/include/torch/csrc/api/include/torch/nn/cloneable.h:69:61: error: invalid static_cast from type ‘const torch::OrderedDict<std::basic_string<char>, std::shared_ptr<torch::nn::Module> >’ to type ‘torch::OrderedDict<std::basic_string<char>, std::shared_ptr<torch::nn::Module> >&’ /disk1/zsxm/miniconda3/envs/pytorch/lib/python3.7/site-packages/torch/include/torch/csrc/api/include/torch/nn/cloneable.h: In instantiation of ‘std::shared_ptr<torch::nn::Module> torch::nn::Cloneable<Derived>::clone(const c10::optional<c10::Device>&) const [with Derived = torch::nn::SequentialImpl]’: /tmp/tmpxft_0000d2f7_00000000-6_basis_cuda.cudafe1.stub.c:51:27: required from here /disk1/zsxm/miniconda3/envs/pytorch/lib/python3.7/site-packages/torch/include/torch/csrc/api/include/torch/nn/cloneable.h:57:59: error: invalid static_cast from type ‘const torch::OrderedDict<std::basic_string<char>, at::Tensor>’ to type ‘torch::OrderedDict<std::basic_string<char>, at::Tensor>&’ /disk1/zsxm/miniconda3/envs/pytorch/lib/python3.7/site-packages/torch/include/torch/csrc/api/include/torch/nn/cloneable.h:69:61: error: invalid static_cast from type ‘const torch::OrderedDict<std::basic_string<char>, std::shared_ptr<torch::nn::Module> >’ to type ‘torch::OrderedDict<std::basic_string<char>, std::shared_ptr<torch::nn::Module> >&’ /disk1/zsxm/miniconda3/envs/pytorch/lib/python3.7/site-packages/torch/include/torch/csrc/api/include/torch/nn/cloneable.h: In instantiation of ‘std::shared_ptr<torch::nn::Module> torch::nn::Cloneable<Derived>::clone(const c10::optional<c10::Device>&) const [with Derived = torch::nn::ModuleListImpl]’: /tmp/tmpxft_0000d2f7_00000000-6_basis_cuda.cudafe1.stub.c:51:27: required from here /disk1/zsxm/miniconda3/envs/pytorch/lib/python3.7/site-packages/torch/include/torch/csrc/api/include/torch/nn/cloneable.h:57:59: error: invalid static_cast from type ‘const torch::OrderedDict<std::basic_string<char>, at::Tensor>’ to type ‘torch::OrderedDict<std::basic_string<char>, at::Tensor>&’ /disk1/zsxm/miniconda3/envs/pytorch/lib/python3.7/site-packages/torch/include/torch/csrc/api/include/torch/nn/cloneable.h:69:61: error: invalid static_cast from type ‘const torch::OrderedDict<std::basic_string<char>, std::shared_ptr<torch::nn::Module> >’ to type ‘torch::OrderedDict<std::basic_string<char>, std::shared_ptr<torch::nn::Module> >&’ /disk1/zsxm/miniconda3/envs/pytorch/lib/python3.7/site-packages/torch/include/torch/csrc/api/include/torch/nn/cloneable.h: In instantiation of ‘std::shared_ptr<torch::nn::Module> torch::nn::Cloneable<Derived>::clone(const c10::optional<c10::Device>&) const [with Derived = torch::nn::ModuleDictImpl]’: /tmp/tmpxft_0000d2f7_00000000-6_basis_cuda.cudafe1.stub.c:51:27: required from here /disk1/zsxm/miniconda3/envs/pytorch/lib/python3.7/site-packages/torch/include/torch/csrc/api/include/torch/nn/cloneable.h:57:59: error: invalid static_cast from type ‘const torch::OrderedDict<std::basic_string<char>, at::Tensor>’ to type ‘torch::OrderedDict<std::basic_string<char>, at::Tensor>&’ /disk1/zsxm/miniconda3/envs/pytorch/lib/python3.7/site-packages/torch/include/torch/csrc/api/include/torch/nn/cloneable.h:69:61: error: invalid static_cast from type ‘const torch::OrderedDict<std::basic_string<char>, std::shared_ptr<torch::nn::Module> >’ to type ‘torch::OrderedDict<std::basic_string<char>, std::shared_ptr<torch::nn::Module> >&’ /disk1/zsxm/miniconda3/envs/pytorch/lib/python3.7/site-packages/torch/include/torch/csrc/api/include/torch/nn/cloneable.h: In instantiation of ‘std::shared_ptr<torch::nn::Module> torch::nn::Cloneable<Derived>::clone(const c10::optional<c10::Device>&) const [with Derived = torch::nn::TransformerDecoderImpl]’: /disk1/zsxm/miniconda3/envs/pytorch/lib/python3.7/site-packages/torch/include/torch/csrc/api/include/torch/optim/sgd.h:49:48: required from here /disk1/zsxm/miniconda3/envs/pytorch/lib/python3.7/site-packages/torch/include/torch/csrc/api/include/torch/nn/cloneable.h:57:59: error: invalid static_cast from type ‘const torch::OrderedDict<std::basic_string<char>, at::Tensor>’ to type ‘torch::OrderedDict<std::basic_string<char>, at::Tensor>&’ /disk1/zsxm/miniconda3/envs/pytorch/lib/python3.7/site-packages/torch/include/torch/csrc/api/include/torch/nn/cloneable.h:69:61: error: invalid static_cast from type ‘const torch::OrderedDict<std::basic_string<char>, std::shared_ptr<torch::nn::Module> >’ to type ‘torch::OrderedDict<std::basic_string<char>, std::shared_ptr<torch::nn::Module> >&’ /disk1/zsxm/miniconda3/envs/pytorch/lib/python3.7/site-packages/torch/include/torch/csrc/api/include/torch/nn/cloneable.h: In instantiation of ‘std::shared_ptr<torch::nn::Module> torch::nn::Cloneable<Derived>::clone(const c10::optional<c10::Device>&) const [with Derived = torch::nn::TransformerEncoderImpl]’: /disk1/zsxm/miniconda3/envs/pytorch/lib/python3.7/site-packag [... cut ...] '"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-0ejz4nxt/install-record.txt --single-version-externally-managed --compile --install-headers /disk1/zsxm/miniconda3/envs/pytorch/include/python3.7m/torch-spline-conv Check the logs for full command output.

StiphyJay, 2022-08-07:
> replace

I solved this bug by change the pytorch code like this:
change the code in anaconda3/envs/pillarnet-torch18/lib/python3.7/site-packages/torch/include/torch/csrc/api/include/torch/nn/cloneable.h:

at line 46/58/70 change the original code : 

copy->parameters_.size() == parameters_.size()

copy->buffers_.size() == buffers_.size()

copy->children_.size() == children_.size()

to

copy->parameters_.size() == this -> parameters_.size()

copy->buffers_.size() == this -> buffers_.size()

copy->children_.size() == this -> children_.size()

Then, It works.

wang-chen, 2022-08-07:
what is the PyTorch version you installed?
Currently, we support ``PyTorch 1.11`` and ``functorch 0.1.1`` (they should not need any changes).

StiphyJay, 2022-08-08:
> 

My env:

functorch                0.1.1
torch                    1.11.0+cu113

closed 2022-08-12 by wang-chen