pypose/pypose #120: Add Bundle Adjustment Module

wang-chen, 2022-08-29:
### 🚀 The feature, motivation and pitch

As stated in the title, a differential bundle adjustment module is required in many SLAM systems.
This module may require the following implemented functionalities.

- [LM optimizer](https://pypose.org/docs/main/generated/pypose.optim.LevenbergMarquardt/)
- [Model Jacobian Computation](https://pypose.org/docs/main/generated/pypose.optim.functional.modjac/)

lizimo061, 2023-04-11:
Hi there, I wrote a small example based on [BAL dataset](https://grail.cs.washington.edu/projects/bal/), but I found it is extremely slow even on the smallest dataset(with the vectorize option off, CUDA out of memory reported if on), any good suggestion? 
`class BAGraph(nn.Module):

    def __init__(self, cameras, points, intrinsics):
        super().__init__()
        self.cameras = pp.Parameter(cameras)
        self.points = nn.Parameter(points)
        self.intrinsics = nn.Parameter(intrinsics)

    def forward(self, observations, camera_index, pt_index):
        pt_cam = self.cameras[camera_index, ...].Act(self.points[pt_index, ...])
        pt_image = -pt_cam[..., :2] / torch.unsqueeze(pt_cam[..., 2], 1)
        pt_norm = torch.linalg.vector_norm(pt_image, ord=2, dim=1)
        distortion_factors = 1 + self.intrinsics[camera_index, ..., 1] * pt_norm**2 + self.intrinsics[camera_index, ...,
                                                                                                      2] * pt_norm**4
        pt_pixel = torch.unsqueeze(self.intrinsics[camera_index, ..., 0], 1) * torch.unsqueeze(distortion_factors,
                                                                                               1) * pt_image
        error = pt_pixel - observations
        return error`

wang-chen, 2023-04-12:
Hi, we are currently implementing sparse tensors, which will solve this problem. @zitongzhan

zitongzhan, 2023-04-17:
BA is dependent on this issue: https://github.com/pypose/pypose/issues/207

hxu296, 2023-07-08:
> Hi there, I wrote a small example based on [BAL dataset](https://grail.cs.washington.edu/projects/bal/), but I found it is extremely slow even on the smallest dataset(with the vectorize option off, [...]

Hello, you can try an unofficial pypose-native BAL colab example [here]( https://colab.research.google.com/github/hxu296/pypose/blob/sparse/ba/examples/sparse/ba_example.ipynb). The smallest problem (problem-49-7776-pre) works fine on both CPU and GPU, and you can change the problem by using CLI parameters.

chenjiajie9811, 2024-06-18:
Any update on the official implementation for the differentiable BA?

zitongzhan, 2024-07-26:
> Any update on the official implementation for the differentiable BA?

We are preparing for it and tuning its final APIs.

jiaming-ai, 2024-10-02:
hi, any expected date for releasing this feature?

oliver-batchelor, 2024-12-06:
"[Bundle Adjustment in the Eager Mode](https://arxiv.org/abs/2409.12190)"  - is this an implementation for this feature in pypose?

zitongzhan, 2024-12-06:
> "[Bundle Adjustment in the Eager Mode](https://arxiv.org/abs/2409.12190)" - is this an implementation for this feature in pypose?

Yes. It will be an implementation mimicking pypose API and sharing many components.

oliver-batchelor, 2024-12-10:
> > "[Bundle Adjustment in the Eager Mode](https://arxiv.org/abs/2409.12190)" - is this an implementation for this feature in pypose?
> 
> Yes. It will be an implementation mimicking pypose API and sharing many components.

Very cool. I think it will get a lot of use.  

Does it exist as a branch or work-in-progress here or will it be a new library which adds this to pypose?