pypose/pypose #126: weight matrix in optimization

wang-chen, 2022-09-08:
### 🐛 Describe the bug

The current implementation for the weight least square problem is to get the Cholesky decomposition of the weight matrix, while it may not be positive-definite.
A better implementation is to get the residual as `R= J.T @ W @ e` and `H = J.T @ W @ J`, which will not need any decomposition.

### Versions

No version needed

Jaraxxus-Me, 2022-09-08:
Does J mean Jacobian Matrix?

Jaraxxus-Me, 2022-09-08:
And what is the use of H = J.T @ W @ J ?
Where is H used ?

wang-chen, 2022-09-08:
Yes, `J` is Jacobian,
-  `J.T @ J` -- > `J.T @ W @ J` refers to this line 
https://github.com/pypose/pypose/blob/44ab958e96bd077639f71d5a86df8c7d1e8ecc9e/pypose/optim/optimizer.py#L455: 

- `J.T @ R` -->  `J.T @ W @ R` refers to this line
https://github.com/pypose/pypose/blob/44ab958e96bd077639f71d5a86df8c7d1e8ecc9e/pypose/optim/optimizer.py#L460

Jaraxxus-Me, 2022-09-08:
Raising this problem for line455[](url)
[image]

closed 2023-01-16 by xukuanHIT