pypose/pypose #158: torch.cuda.OutOfMemoryError: CUDA out of memory while run pgo example

jay80253504, 2022-11-29:
### 🐛 Describe the bug

hi, thank you for the great job!
when i run pgo.py as follow:
```
python examples/module/pgo/pgo.py --dataroot /home/data/pypose/pgo/ --device cuda:0 --radius 1000
```
i got such error:
```
torch.cuda.OutOfMemoryError: CUDA out of memory. Tried to allocate 31.69 GiB (GPU 0; 31.75 GiB total capacity; 5.28 GiB already allocated; 25.38 GiB free; 5.29 GiB reserved in total by PyTorch) If reserved memory is >> allocated memory try setting max_split_size_mb to avoid fragmentation.  See documentation for Memory Management and PYTORCH_CUDA_ALLOC_CONF
```
 i'm confused why need 31.69 GiB memory? is there any way to solve this? it's too large for my poor device.

### Versions

my environment information is as follow:

PyTorch version: 1.13.0+cu116
Is debug build: False
CUDA used to build PyTorch: 11.6
ROCM used to build PyTorch: N/A

OS: Ubuntu 18.04.6 LTS (x86_64)
GCC version: (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0
Clang version: Could not collect
CMake version: version 3.10.2
Libc version: glibc-2.27

Python version: 3.8.15 | packaged by conda-forge | (default, Nov 22 2022, 08:49:35)  [GCC 10.4.0] (64-bit runtime)
Python platform: Linux-3.10.0-957.27.2.el7.x86_64-x86_64-with-glibc2.10
Is CUDA available: True
CUDA runtime version: 11.6.124
GPU models and configuration: 
GPU 0: Tesla V100-SXM2-32GB
GPU 1: Tesla V100-SXM2-32GB

Nvidia driver version: 450.80.02
cuDNN version: Probably one of the following:
/usr/lib/x86_64-linux-gnu/libcudnn.so.8.4.0
/usr/lib/x86_64-linux-gnu/libcudnn_adv_infer.so.8.4.0
/usr/lib/x86_64-linux-gnu/libcudnn_adv_train.so.8.4.0
/usr/lib/x86_64-linux-gnu/libcudnn_cnn_infer.so.8.4.0
/usr/lib/x86_64-linux-gnu/libcudnn_cnn_train.so.8.4.0
/usr/lib/x86_64-linux-gnu/libcudnn_ops_infer.so.8.4.0
/usr/lib/x86_64-linux-gnu/libcudnn_ops_train.so.8.4.0
HIP runtime version: N/A
MIOpen runtime version: N/A
Is XNNPACK available: True

Versions of relevant libraries:
[pip3] numpy==1.23.4
[pip3] torch==1.13.0+cu116
[pip3] torchaudio==0.13.0+cu116
[pip3] torchvision==0.14.0+cu116
[conda] numpy                     1.23.4                   pypi_0    pypi
[conda] torch                     1.13.0+cu116             pypi_0    pypi
[conda] torchaudio                0.13.0+cu116             pypi_0    pypi
[conda] torchvision               0.14.0+cu116             pypi_0    pypi

wang-chen, 2022-11-30:
Hi currently you may run with the following options to save memory or acceleration:
```bash
    --no-vectorize       to save memory
    --vectorize          to accelerate computation
```
For example:
```
python pgo.py --dataroot DATAROOT --device cuda:0 --no-vectorize
```

The current memory consumption is large because we haven't integrate sparse block tensors into the optimization. We are currently working on this and will have it in the next major release. However, the interface (API) will be mostly unchanged, you may use it safely now.

jay80253504, 2022-12-01:
> Hi currently you may run with the following options to save memory or acceleration:
> 
> ```shell
>     --no-vectorize       to save memory
>     --vectorize          to accelerate computation
> ```
> 
> For example:
> 
> ```
> python pgo.py --dataroot DATAROOT --device cuda:0 --no-vectorize
> ```
> 
> The current memory consumption is large because we haven't integrate sparse block tensors into the optimization. We are currently working on this and will have it in the next major release. However, the interface (API) will be mostly unchanged, you may use it safely now.

Hi, thank you for the reply!

i add ```--no-vectorize``` and the question is solved.

closed 2022-12-01 by jay80253504