pypose/pypose #181: Automate release and documentation upload

Xanthorapedia, 2023-02-03:
## Summary

We would like a mechanism for automatically updating [pypose.org/docs](pypose.org/docs) that:

1. Centralizes artifact building (downloads built artifacts from CircleCI instead of building again as in [the current workflow](https://github.com/pypose/pypose/blob/f285966a027dd392826dfad8d4696eaf34589af4/.github/workflows/python-publish.yml#L33).
2. Pulls the latest release version and upload it to [pypose/pypose.github.io/docs](https://github.com/pypose/pypose.github.io/tree/main/docs
).
3. Updates [pypose/pypose.github.io/docs/versions.html](https://github.com/pypose/pypose.github.io/tree/main/docs/versions.html) to reflect the newly added doc.

## Improvements

More consistent artifact building. No need to manually upload and update docs page.

## Risks

No known risks.

## Involved components

[.github/workflows/python-publish.yml](https://github.com/pypose/pypose/tree/main/.github/workflows/python-publish.yml)
[.circleci/config.yml](https://github.com/pypose/pypose/tree/main/.circleci/config.yml)

zitongzhan, 2023-02-15:
Maybe take a look at [mmdet](https://github.com/open-mmlab/mmdetection). They implemented similar mechanisms.

wang-chen, 2023-02-15:
@zitongzhan Just checked mmdet, they have a similar doc template, but it seems that they don't implement the auto doc versioning. We want something like this: https://pypose.org/docs/versions.html
The current doc v0.2.2 was uploaded manually but we want to make it automatic when new versions are released via [this action](https://github.com/pypose/pypose/actions/workflows/python-publish.yml). Please correct me if I am wrong.

zitongzhan, 2023-02-15:
> @zitongzhan Just checked mmdet, they have a similar doc template, but it seems that they don't implement the auto doc versioning. We want something like this: https://pypose.org/docs/versions.html [...]

I thought they had that since their document version always synchronize with package release. I will take a closer look on this.
[image]

wang-chen, 2023-02-16:
oh yes, I didn't notice that they put the versioning in the left bottom corner. Thank you!

zitongzhan, 2023-02-17:
I found out that the version inside mmdet's document is not part of their compiled html. Instead, it is manage by this [service](https://readthedocs.org).

closed 2024-08-13 by zitongzhan