pypose/pypose #196: PyPose SO3 API request

haleqiu, 2023-03-11:
### 🚀 The feature, motivation and pitch

The current lie tensor API will not follow the data type of the input numpy array.
```
>>> x = np.array([0.,0.,0.,1.], dtype =np.float64)
>>> x.dtype
dtype('float64')
>>> pp.SO3(x).dtype
torch.float32
```
Giving that PyTorch supports the function. I suggest that we should also follow this API

wang-chen, 2023-03-11:
This request seems not reasonable. Here is output from PyTorch.
```python
>>> import torch, pypose as pp, numpy as np
>>> x = np.array([0.,0.,0.,1.], dtype=np.float64)
>>> y = torch.Tensor(x)
>>> print(x.dtype, y.dtype)
float64 torch.float32
```

haleqiu, 2023-03-11:
It depends on the which `torch` API you are using

```
>>> x = np.array([0.,0.,0.,1.], dtype=np.float64)
>>> y = torch.tensor(x)
>>> print(x.dtype, y.dtype)
float64 torch.float64
```

wang-chen, 2023-03-11:
`SO3` is an alias of `LieTensor`.
Suggest to run
```python
>>> x = torch.from_numpy(np.array([0.,0.,0.,1.], dtype =np.float64))
>>> y = pp.SO3(x)
>>> print(y.dtype)
torch.float64
```

closed 2023-04-06 by zitongzhan