pypose/pypose #207: Sparse Block Tensor: finalize the current design and add examples

huyaoyu, 2023-03-29:
Finalize the current design for Sparse Block Tensor and add new examples. This makes it easier for Zitong to start collaborating.

zitongzhan, 2023-04-12:
Branched out to `zitong_sparse ` to continue development, in which main branch update has been merged: https://github.com/pypose/pypose/tree/zitong_sparse

zitongzhan, 2023-04-13:
Today's commit made SBT torch.jit compatible and thus faster. Here's a speed comparison
when running a SBT @ SBT operation 10,000 times. 
```
@zitongzhan ➜ /workspaces/pypose (zitong_sparse) $ PYTORCH_JIT=1 /usr/local/bin/python /workspaces/pypose/test/sparse/test_basic.py
elapsed time = 7.679962396621704
@zitongzhan ➜ /workspaces/pypose (zitong_sparse) $ PYTORCH_JIT=0 /usr/local/bin/python /workspaces/pypose/test/sparse/test_basic.py
elapsed time = 10.87170696258545
```
Without JIT, it consumes 40% more time to complete!
```python
>>> (10.87170696258545 - 7.679962396621704) / 7.679962396621704
0.4155937752205328
```

zitongzhan, 2023-04-17:
TODOs:

- [ ] JIT: https://github.com/pypose/pypose/pull/222
- [ ] pytest factory for all operations
- [ ] The basic Linear Algebra operations 
- [ ] Reformat for code review

zitongzhan, 2023-04-27:
Simplify logic inside coo2hybrid to further speed up `@` operation
Speed is tested by the snippet
```
if __name__ == '__main__':
    i = [[0, 0, 1, 2],[0, 2, 1, 2]]
    v = torch.arange(16).view((4, 2, 2)).to(dtype=torch.float32)
    x = sparse_block_tensor(i, v, size=(3, 3), dtype=torch.float32)

    import time
    start = time.time()
    for i in range(10000):
        x @ x
    end = time.time()
    print(f'elapsed time = {end - start}')
```
Before the change:
Three runs:
```
@zitongzhan ➜ /workspaces/pypose (SparseTensorBlock) $ /usr/local/bin/python 
elapsed time = 6.4026172161102295
@zitongzhan ➜ /workspaces/pypose (SparseTensorBlock) $ /usr/local/bin/python 
elapsed time = 6.1571571826934814
@zitongzhan ➜ /workspaces/pypose (SparseTensorBlock) $ /usr/local/bin/python 
elapsed time = 6.2495129108428955
```
After the change:
```
@zitongzhan ➜ /workspaces/pypose (SparseTensorBlock) $ /usr/local/bin/python 
elapsed time = 4.711676597595215
@zitongzhan ➜ /workspaces/pypose (SparseTensorBlock) $ /usr/local/bin/python 
elapsed time = 4.941352367401123
@zitongzhan ➜ /workspaces/pypose (SparseTensorBlock) $ /usr/local/bin/python 
elapsed time = 4.785998344421387
```
Waiting for a commit link.

zitongzhan, 2023-04-29:
A runtime graph associated with the above PR
[image]

closed 2024-01-05 by zitongzhan