pypose/pypose #231: Optimization in `se3.Exp` and `vec2skew`

NicolayP, 2023-04-27:
### 🚀 The feature, motivation and pitch

Optimizing the two operations. The two operations (only one I need so haven't check for others) create a padding, An eye matrix for exp and 0  diagonal for skew. Those are then uploaded on the device when cuda is enabled. 

Use case: 
I have a rigid body step model that need to be called recursively to propagate many samples. Subsequent calls to `se3.exp` leads to a high number of memcpy instructions drastically increasing the computation time. `vec2skew` is also called in this function.

### Alternatives

With a sligh memory footprint, one could  convert `vec2skew` and `se3.exp` into singletons object that can live in GPU memory with all constants already allocated on the device at creation time. Avoiding the need to call memcpy every time those instruction are being called.

Would that be something that can be added/will be added?

wang-chen, 2023-04-27:
Hi @NicolayP, do you mean change `vec2skew` to something like
```python
def vec2skew(input:torch.Tensor) -> torch.Tensor:
    v = input.tensor() if hasattr(input, 'ltype') else input
    assert v.shape[-1] == 3, "Last dim should be 3"
    skew = torch.zeros(v.shape + (3,), device=v.device, dtype=v.dtype)
    skew[..., 0, 1], skew[..., 0, 2] = -v[..., 2],   v[..., 1]
    skew[..., 1, 0], skew[..., 1, 2] =  v[..., 2],  -v[..., 0]
    skew[..., 2, 0], skew[..., 2, 1] = -v[..., 1],   v[..., 0]
    return skew
```
instead of 
```python
    v = input.tensor() if hasattr(input, 'ltype') else input
    assert v.shape[-1] == 3, "Last dim should be 3"
    O = torch.zeros(v.shape[:-1], device=v.device, dtype=v.dtype, requires_grad=v.requires_grad)
    return torch.stack([torch.stack([        O, -v[...,2],  v[...,1]], dim=-1),
                        torch.stack([ v[...,2],         O, -v[...,0]], dim=-1),
                        torch.stack([-v[...,1],  v[...,0],         O], dim=-1)], dim=-2)
```

NicolayP, 2023-04-27:
Hi @wang-chen not really, this is the same problem I was thinking at something like this: 
```python
...
skew = torch.zeros((1, 3, 3), dtype=torch.havent_figured_that_one_out_yet)
...
def to(device):
   skew.to(device)
...
def vec2skew(input:torch.Tensor) -> torch.Tensor:
    v = input.tensor() if hasattr(input, 'ltype') else input
    assert v.shape[-1] == 3, "Last dim should be 3"
    skew_br = torch.broadcast_to(skew, v.shape + (3,)) # already living on device (cuda)
    skew_br[..., 0, 1], skew_br[..., 0, 2] = -v[..., 2],   v[..., 1]
    skew_br[..., 1, 0], skew_br[..., 1, 2] =  v[..., 2],  -v[..., 0]
    skew_br[..., 2, 0], skew_br[..., 2, 1] = -v[..., 1],   v[..., 0]
    return skew_br
```

obviously this is very nasty. another solution is to make skew a singleton something like:
ething like this: 
```python
...
class vec2skew(torch.nn.Module):
   def __init__(self, dtype):
      self.register_buffer("skew", torch.zeros(shape=(1, 3, 3), dtype)

   def forward(self, input:torch.Tensor) -> torch.Tensor:
        v = input.tensor() if hasattr(input, 'ltype') else input
        assert v.shape[-1] == 3, "Last dim should be 3"
        skew = torch.broadcast_to(self.skew, v.shape + (3,)) # already living on device (cuda)
        skew[..., 0, 1], skew[..., 0, 2] = -v[..., 2],   v[..., 1]
        skew[..., 1, 0], skew[..., 1, 2] =  v[..., 2],  -v[..., 0]
        skew[..., 2, 0], skew[..., 2, 1] = -v[..., 1],   v[..., 0]
        return skew
```
then create a skew object that can be persistent in GPU memory and will be called when `vec2skew` is called.

obviously this requires way more work and Idk if it will conflict with other parts of the library software architecture.

wang-chen, 2023-04-27:
Hi @NicolayP , I guess this is not a good solution, since

1. the users may have in-place operation on the zero values of the returned skew matrix, which will raise potential bugs; on the other hand, if the in-place operation cannot modify the zero tensors, it means it copies the zeros tensors, which will have the same effect of the current implementation. 
2. transfer data between `cpu` and `gpu` is very time-consuming, thus we don't suggest call `tensor.to(device)`, instead we call `torch.tensor(..., device=device)` to save time.

wang-chen, 2023-04-27:
the suggested commodification below did avoid creating the small zero tensor of `O = torch.zeros(v.shape[:-1], ...)` defined in the original implementation.
```python
    v = input.tensor() if hasattr(input, 'ltype') else input
    assert v.shape[-1] == 3, "Last dim should be 3"
    O = torch.zeros(v.shape[:-1], device=v.device, dtype=v.dtype, requires_grad=v.requires_grad)
    return torch.stack([torch.stack([        O, -v[...,2],  v[...,1]], dim=-1),
                        torch.stack([ v[...,2],         O, -v[...,0]], dim=-1),
                        torch.stack([-v[...,1],  v[...,0],         O], dim=-1)], dim=-2)
```

closed 2023-05-03 by NicolayP