pypose/pypose #241: SE3_Adj and SO3_Adj  function error

Lutonin, 2023-05-11:
### 🐛 Describe the bug

## Description
Im trying to calculate left-invariant error according to https://arxiv.org/pdf/1812.01537 (eq. 54)
I realized that there is no other way to calculate pure adjoint of Lie group except using SE3_Adj(X) function from operation module
But this function starts with error  "RuntimeError: linalg.cross: inputs dimension -1 must have length 3. Got 0 and 3"
I realized that if change lines
```python
Xw_3x3 = Xw.unsqueeze(-1) * I3x3
```
in SO3_Adj(X) module  (line 174 in operation.py) to 
```python
Xw_3x3 = I3x3 * Xw.unsqueeze(-1) 
```
all starts to work fine 

## Steps to reproduse
```python
import pypose as pp
X = pp.randn_SE3()
pp.lietensor.operation.SE3_Adj(X)
```
## PS
If  there is an another way to calulate left-invariant error without matrix representation, please help me find it out .

### Versions

Collecting environment information...
PyTorch version: 2.0.1+cu117
Is debug build: False
CUDA used to build PyTorch: 11.7
ROCM used to build PyTorch: N/A

OS: Debian GNU/Linux 10 (buster) (x86_64)
GCC version: (Debian 8.3.0-6) 8.3.0
Clang version: Could not collect
CMake version: version 3.26.3
Libc version: glibc-2.28

Python version: 3.10.11 (main, May  3 2023, 09:31:23) [GCC 8.3.0] (64-bit runtime)
Python platform: Linux-5.4.209-116.367.amzn2.x86_64-x86_64-with-glibc2.28
Is CUDA available: False
CUDA runtime version: No CUDA
GPU models and configuration: No CUDA
Nvidia driver version: No CUDA
cuDNN version: No CUDA
HIP runtime version: N/A
MIOpen runtime version: N/A
Is XNNPACK available: True

Versions of relevant libraries:
[pip3] numpy==1.23.4
[pip3] torch==2.0.1
[pip3] torchvision==0.13.1
[conda] Could not collect

wang-chen, 2023-05-11:
hi do you mean you want this [Adj](https://pypose.org/docs/main/generated/pypose.Adj/?highlight=adj#pypose.Adj) function?

wang-chen, 2023-05-11:
BTW, @Lutonin, `pp.lietensor.operation` is not designed to expose to users. You may use the search box in our [documentation](https://pypose.org/docs/main/functions/) to find API designed to expose to users.

wang-chen, 2023-05-13:
@Lutonin Did this [Adj](https://pypose.org/docs/main/generated/pypose.Adj/?highlight=adj#pypose.Adj) function solve your question? We would like to provide more assistance or close this issue.

closed 2023-05-24 by wang-chen