pypose/pypose #242: PGO for several agents

VladimirYugay, 2023-05-15:
I have 2 sequences of transformations `T = (R, t)` in homogeneous coordinates of shape `(n, 4, 4)`. 

I know the exact transformations between the sequences' first and last pairs. 

I want to optimize the rest of the poses using the constraints I've mentioned above. 

However, I don't really understand the example you provided for PGO. Particularly:

1. What is the format and why do I need info in the G2OPGO dataset?
2. What is the edge format that you expect? I've checked the repo for the garage dataset but found only the description of the vertex. 
3. Why does the error function in the PGO example look like this: `error = poses.Inv() @ node1.Inv() @ node2`
4. How might the error function look in my specific case?

wang-chen, 2023-05-15:
1. What is the format and why do I need info in the G2OPGO dataset?
- I guess you mean the information matrix, it defines the weights (or covariance) of your measurements. You should consider that for every optimization problem. If you don't know the covariance, set them as identity matrix or some pre-defined values could be fine.

2. What is the edge format that you expect? I've checked the repo for the garage dataset but found only the description of the vertex.
- the module can be any format, just output a tensor in the forward function. The PGO dataset/file has edge definition in the lower part of the file, scroll down the file you will see that.

3. Why does the error function in the PGO example look like this: `error = poses.Inv() @ node1.Inv() @ node2`
- It is the definition of factor graphs, i.e., the measurement between two poses (`poses`) should be as close as possible to their pose difference `node1.Inv() @ node2`
- the module forward function just outputs anything you want to be close to zero.

4. How might the error function look in my specific case?
- To optimize the rest of the poses using the constraints, just don't define those variables as `nn.Parameter` (for Tensor) or `pypose.Parameter` (for LieTensor). The principle is to only define those variables you want to optimize as `Parameter`.

Since you are using homogeneous coordinates, here is some functions you may need:
- [homo2cart](https://pypose.org/docs/main/generated/pypose.homo2cart/#pypose.homo2cart)
- [cart2homo](https://pypose.org/docs/main/generated/pypose.cart2homo/#pypose.cart2homo)

closed 2023-05-24 by wang-chen

VladimirYugay, 2023-06-30:
Thanks for your response. Regarding freezing the nodes. For my specific use case, it's not very convenient to wrap and unwrap the node's parameters. I was wondering if I remove all the ongoing and outgoing edges from a specific node, will it be optimized? In theory, it should stay the same, or?