pypose/pypose #247: Return type of `mul()` is not consistent with the documentation

MarkChenYutian, 2023-05-30:
### 🐛 Describe the bug

`mul` function's behavior is not aligned with the [documentation](https://pypose.org/docs/main/generated/pypose.mul/?highlight=mul#pypose.mul).

```
>>> import pypose as pp
>>> I = pp.identity_SE3()
>>> I.mul(I)
tensor([0., 0., 0., 0., 0., 0., 1.])
>>> type(I.mul(I))
<class 'torch.Tensor'>
```

By documentation, `mul` function should return `LieGroup` when `input` and `other` are of type `LieGroup`.

### Versions

```
Collecting environment information...
PyTorch version: 2.0.1+cu117
Is debug build: False
CUDA used to build PyTorch: 11.7
ROCM used to build PyTorch: N/A

OS: Ubuntu 18.04.6 LTS (x86_64)
GCC version: (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0
Clang version: Could not collect
CMake version: version 3.26.3
Libc version: glibc-2.27

Python version: 3.8.16 (default, Mar  2 2023, 03:21:46)  [GCC 11.2.0] (64-bit runtime)
Python platform: Linux-5.15.90.1-microsoft-standard-WSL2-x86_64-with-glibc2.17
Is CUDA available: True
CUDA runtime version: 11.0.221
GPU models and configuration: GPU 0: NVIDIA GeForce RTX 3060 Laptop GPU
Nvidia driver version: 516.59
cuDNN version: Probably one of the following:
/usr/lib/x86_64-linux-gnu/libcudnn.so.8.0.5
/usr/lib/x86_64-linux-gnu/libcudnn_adv_infer.so.8.0.5
/usr/lib/x86_64-linux-gnu/libcudnn_adv_train.so.8.0.5
/usr/lib/x86_64-linux-gnu/libcudnn_cnn_infer.so.8.0.5
/usr/lib/x86_64-linux-gnu/libcudnn_cnn_train.so.8.0.5
/usr/lib/x86_64-linux-gnu/libcudnn_ops_infer.so.8.0.5
/usr/lib/x86_64-linux-gnu/libcudnn_ops_train.so.8.0.5
HIP runtime version: N/A
MIOpen runtime version: N/A
Is XNNPACK available: True

Versions of relevant libraries:
[pip3] numpy==1.22.3
[pip3] torch==2.0.1
[conda] cudatoolkit               10.2.89             h713d32c_10    conda-forge
[conda] mkl                       2023.1.0         h6d00ec8_46342  
[conda] mkl-service               2.4.0            py38h5eee18b_1  
[conda] numpy                     1.22.3           py38h99721a1_2    conda-forge
[conda] torch                     2.0.1                    pypi_0    pypi
```

wang-chen, 2023-05-31:
Hi, thanks for creating this issue. For better readability, we suggest calling ``a * b`` or ``a @ b`` instead of ``a.mul(b)`` or ``pypose.mul(a, b)`` .  They have the same behaviors in the new PR.

MarkChenYutian, 2023-05-31:
Got it, thanks for replying!

closed 2023-05-31 by MarkChenYutian