pypose/pypose #252: broadcast_inputs triggers tensor storange copy, peaks CUDA memory consumption

hxu296, 2023-06-26:
## Summary

It seems that the following line in `def broadcast_inputs(x, y)` triggerred a tensor storage copy that caused a CUDA memory overflow when I tried to run a small bundle adjustment dataset with 31843 observations. Both `reshape` and `contiguous` could trigger a memory copy. If we can avoid memory copy in `broadcast_inputs`, we can potentially avoid overflowing CUDA memory at this step.

https://github.com/pypose/pypose/blob/6598a84c8fb993a1825ee9629df035a985111312/pypose/lietensor/operation.py#L914

[image]

## Improvements

refactor broadcast_inputs to not use reshape and contiguous.

## Risks

TBD

## Involved components

- [pypose/pypose/lietensor/operation.py](https://github.com/pypose/pypose/blob/6598a84c8fb993a1825ee9629df035a985111312/pypose/lietensor/operation.py#L914)

## Optional: Intended side effects

TBD

## Optional: Missing test coverage

TBD

wang-chen, 2023-06-30:
@hxu296 This is a historical issue, the functions were implemented by Cuda and the tensor copy behavior was expected. But now it seems not needed anymore. A similar function could be `torch.broadcast_tensors` and `torch.broadcast_shapes`, you may re-check those functions called after `broadcast_inputs`, if they are implemented by batch, then `broadcast_inputs` can be safely removed.

cola1917, 2026-10-24:
I'd like to work on this. The copy is still present on main 9345958: broadcasting one float32 SE3 pose over 262,144 points allocates a 7 MiB pose copy on CUDA.

I'll check all broadcast_inputs callers and preserve their batch dimensions as expanded views where the kernels support them, with forward/backward equivalence tests for the four Lie groups, non-contiguous/multidimensional inputs, and an end-to-end CUDA memory benchmark. This will stay separate from the tangent-gradient API work in #407.

Please flag any scope or ownership conflict. Implementation and validation will be AI-assisted. The measurement above reproduces the input copy, not the original complete BA out-of-memory case.

closed 2026-10-25 by wang-chen