pypose/pypose #256: Unexpected Discrepancy of Behavior when Optimizing With/Without Kernel

MarkChenYutian, 2023-07-11:
### 🐛 Describe the bug

I defined a module that output the error as a `Tensor` with dimension of `2 * N * 1` and tried to optimize the model using `LM` optimizer with configurations:

```python
solver    = Cholesky()
strategy  = TrustRegion(radius=1e3)
kernel    = Huber()
optimizer = LM(target, solver=solver, strategy=strategy, kernel=kernel, min=1e-8, vectorize=True, reject=128)
scheduler = StopOnPlateau(optimizer, steps=25, patience=4, decreasing=1e-6, verbose=True)

...

while scheduler.continual():
    # target's `forward` method output an error tensor of shape 2 * target.N * 1
    loss = optimizer.step(input=(), weight=torch.eye(target.N).to("cuda"))
    scheduler.step(loss)
```

The code above executes successfully.
However, when I tried to remove the Huber kernel from optimizer (i.e. change the configuration to following)

```python
solver    = Cholesky()
strategy  = TrustRegion(radius=1e3)
optimizer = LM(target, solver=solver, strategy=strategy, min=1e-8, vectorize=True, reject=128)
scheduler = StopOnPlateau(optimizer, steps=25, patience=4, decreasing=1e-6, verbose=True)

...

while scheduler.continual():
    # target's `forward` method output an error tensor of shape 2 * target.N * 1
    loss = optimizer.step(input=(), weight=torch.eye(target.N).to("cuda"))
    scheduler.step(loss)
```

Following error occurs:

```
Traceback (most recent call last):
  File "/home/yuhengq/miniconda/envs/pypose/lib/python3.8/site-packages/torch/optim/optimizer.py", line 280, in wrapper
    out = func(*args, **kwargs)
  File "/home/yuhengq/miniconda/envs/pypose/lib/python3.8/site-packages/torch/utils/_contextlib.py", line 115, in decorate_context
    return func(*args, **kwargs)
  File "/home/yuhengq/miniconda/envs/pypose/lib/python3.8/site-packages/pypose/optim/optimizer.py", line 445, in step
    J_T = (J_T.unsqueeze(-2) @ weight).squeeze(-2)
RuntimeError: mat1 and mat2 shapes cannot be multiplied (19716x1 and 93x93)
```

After stepping into the optimizer module, I found that in `pypose/optim/optimizer.py`, when using a non-trivial kernel, the model's residual `R` has dimension of `2 * target.N`.

When using a trivial kernel (not providing kernel to optimizer explicitly), the model's residual `R` has dimension of `2 * target.N * 1`. This discrepancy of behavior w/wo the kernel causes the unexpected error above.

https://github.com/pypose/pypose/blob/3b3149208129d2747e779d13cdba3695c1107e62/pypose/optim/optimizer.py#L437-L443

### Versions

```
Collecting environment information...
PyTorch version: 2.0.1+cu117
Is debug build: False
CUDA used to build PyTorch: 11.7
ROCM used to build PyTorch: N/A

OS: Ubuntu 18.04.6 LTS (x86_64)
GCC version: (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0
Clang version: Could not collect
CMake version: version 3.26.3
Libc version: glibc-2.27

Python version: 3.8.16 (default, Mar  2 2023, 03:21:46)  [GCC 11.2.0] (64-bit runtime)
Python platform: Linux-5.15.90.1-microsoft-standard-WSL2-x86_64-with-glibc2.17
Is CUDA available: True
CUDA runtime version: 11.0.221
GPU models and configuration: GPU 0: NVIDIA GeForce RTX 3060 Laptop GPU
Nvidia driver version: 516.59
cuDNN version: Probably one of the following:
/usr/lib/x86_64-linux-gnu/libcudnn.so.8.0.5
/usr/lib/x86_64-linux-gnu/libcudnn_adv_infer.so.8.0.5
/usr/lib/x86_64-linux-gnu/libcudnn_adv_train.so.8.0.5
/usr/lib/x86_64-linux-gnu/libcudnn_cnn_infer.so.8.0.5
/usr/lib/x86_64-linux-gnu/libcudnn_cnn_train.so.8.0.5
/usr/lib/x86_64-linux-gnu/libcudnn_ops_infer.so.8.0.5
/usr/lib/x86_64-linux-gnu/libcudnn_ops_train.so.8.0.5
HIP runtime version: N/A
MIOpen runtime version: N/A
Is XNNPACK available: True

Versions of relevant libraries:
[pip3] numpy==1.22.3
[pip3] torch==2.0.1
[conda] cudatoolkit               10.2.89             h713d32c_10    conda-forge
[conda] mkl                       2023.1.0         h6d00ec8_46342  
[conda] mkl-service               2.4.0            py38h5eee18b_1  
[conda] numpy                     1.22.3           py38h99721a1_2    conda-forge
[conda] torch                     2.0.1                    pypi_0    pypi
```

MarkChenYutian, 2023-07-11:
@haleqiu Could you take a look at this issue?

wang-chen, 2023-07-13:
@MarkChenYutian You need to call a corrector when using a kernel. Your `weight` is N x N matrix, thus your model output has to be (*, N) as well. I suggest using the [FastTriggs](https://pypose.org/docs/main/generated/pypose.optim.corrector.FastTriggs/) corrector.

wang-chen, 2023-07-13:
> @MarkChenYutian You need to call a corrector when using a kernel. Your `weight` is N x N matrix, thus your model output has to be (*, N) as well. I suggest using the [FastTriggs](https://pypose.org/docs/main/generated/pypose.optim.corrector.FastTriggs/) corrector.

The codes run successfully because you didn't pass a corrector to it, so the optimizer calls auto-correction, but the performance if auto-correction is not well, we plan to remove auto-correction in the future.

wang-chen, 2023-07-13:
See this [example](https://github.com/pypose/pypose/blob/3b3149208129d2747e779d13cdba3695c1107e62/tests/optim/test_optimizer.py#L142) for using a corrector.

MarkChenYutian, 2023-07-14:
Got it, thanks! I'll look into the docs for corrector.

closed 2023-07-14 by MarkChenYutian