pypose/pypose #258: LieTensor becomes torch.Tensor in DataParrallel

zeroAska, 2023-07-14:
### 🐛 Describe the bug

I'm trying to use run `DataParrallel` for mulit-gpu training with `pypose's LieTensor`. However, I noticed that when I sent a `LieTensor` pose object to the data parrallel model, the pose becomes a `torch.Tensor`:

Example:
```                                                                                                                                      
import torch                                                                                                                              
import torch.nn                                                                                                                           
import pypose as pp                                                                                                                       
class DataParallelModel(nn.Module):                                                                                                       
    def __init__(self):                                                                                                                   
        super().__init__()                                                                                                                
                                                                                                                                          
    def forward(self, x, T):                                                                                                              
        assert(isinstance(T, pp.LieTensor))                                                                                               
        return T @ x                                                                                                                      
                                                                                                                                          
def test_parallel():                                                                                                                      
    a = pp.randn_SO3(4)                                                                                                                   
    a.requires_grad = True                                                                                                                
                                                                                                                                          
    net = DataParallelModel()                                                                                                             
    if torch.cuda.device_count() > 1:                                                                                                     
        print("Let's use", torch.cuda.device_count(), "GPUs!")                                                                            
        net = nn.DataParallel(net).cuda()                                                                                                 
        net(torch.rand(4), a)                                                                                                             
                                                                                                                                          
test_parallel()
```

Output:
```
Original Traceback (most recent call last):
  File ".../python3.7/site-packages/torch/nn/parallel/parallel_apply.py", line 64, in _wo
rker
    output = module(*input, **kwargs)
  File ".../python3.7/site-packages/torch/nn/modules/module.py", line 1194, in _call_impl
    return forward_call(*input, **kwargs)
  File "test_backprop.py", line 65, in forward
    assert(isinstance(T, pp.LieTensor))
AssertionError  
```

### Versions

```
% python collect_env.py 
Collecting environment information...
PyTorch version: 1.13.1+cu117
Is debug build: False
CUDA used to build PyTorch: 11.7
ROCM used to build PyTorch: N/A

GCC version: (GCC) 7.3.1 
Clang version: Could not collect
CMake version: Could not collect
Libc version: glibc-2.2.5

Python version: 3.7.16
Is CUDA available: True
CUDA runtime version: Could not collect
GPU models and configuration: 
GPU 0: Tesla V100-SXM2-16GB
GPU 1: Tesla V100-SXM2-16GB
GPU 2: Tesla V100-SXM2-16GB
GPU 3: Tesla V100-SXM2-16GB

Nvidia driver version: 515.105.01
cuDNN version: Could not collect
HIP runtime version: N/A
MIOpen runtime version: N/A
Is XNNPACK available: True

Versions of relevant libraries:
[pip3] lietorch==0.4
[pip3] numpy==1.21.6
[pip3] numpydoc==1.5.0
[pip3] torch==1.13.1
[pip3] torch-vision==0.1.6.dev0
[conda] Could not collect
```

wang-chen, 2023-07-14:
This could be a bug from PyTorch DataParrallel which calls [`scatter`](https://github.com/pytorch/pytorch/blob/4328138c1e8c90645fd68235299e0eb7f6d28cfd/torch/nn/parallel/comm.py#L189) function, which could take tensor subclass as inputs, but cannot return tensor subclasses (LieTensor is a subclass of Tensor), especially considering that other torch tensor functions work well for tensor subclass.

We could submit an issue to Pytorch, but that will take a long time. An instant solution is to explicitly convert to LieTensor again if using DataParrallel, e.g.,
```Python
class DataParallelModel(nn.Module):                                                                                                       
    def __init__(self):                                                                                                                   
        super().__init__()                                                                                                                
                                                                                                                                          
    def forward(self, x, T): 
        T = pp.SO3(T)
        return T @ x 
```
BTW, you could use `a = pp.randn_SO3(4, requires_grad=True)` instead of `a = pp.randn_SO3(4)` then `a.requires_grad = True`

zeroAska, 2023-07-14:
Thanks for the response! I added `T = pp.SO3(T)` as follows
```
class DataParallelModel(nn.Module):                                                                                                       
    def __init__(self):                                                                                                                   
        super().__init__()                                                                                                                
                                                                                                                                          
    def forward(self, x, T):          
        print("type of T is ",type(T))                                                                                                    
        assert(isinstance(T, pp.LieTensor))   
        assert(T.is_leaf)                                                                                            
        return T @ x                                                                                                                      
```

but would it to the newly declared `T` being a non-leaf (and thus not optimizable)?
```
  File ".../python3.7/site-packages/torch/nn/modules/module.py", line 1194, in _call_impl
    return forward_call(*input, **kwargs)                                                                                                 
  File "model/test/test_backprop.py", line 68, in forward                                                                                 
    assert(T.is_leaf)                                                                                                                     
AssertionError
```

wang-chen, 2023-07-14:
Yes, it is expected. You create the tensor from `a = randn_SO3()`, so only `a` should be a leaf tensor. If you want something to be optimizable in the module, you need to define it as `pp.Parameter(something)`.

zeroAska, 2023-07-25:
Just to make sure I understood correctly, if I init a LieTensor from an existing leaf tensor, the newly initialized LieTensor is expected to be a non-leaf?

```
        a = torch.tensor([1,2.0,3,4],requires_grad= True)                                                                                                                                                                              
        assert(a.is_leaf)              # True                                                                                                                                                                                                  
        b = (pp.SO3(a))                                                                                                                                                                                                                
        assert(b.is_leaf)              # False                                                                                                                                                                                                
```

zeroAska, 2023-07-27:
Or, in other words, is there a recommended way to convert a non-leaf newly constructed `LieTensor` to a leaf?

wang-chen, 2023-07-27:
Hi @zeroAska, you can easily propagate the grad. See the following example. Note that in the first case, `is_leaf` is `False`, but in the second case, `is_leaf` is `True`.
```
>>> c = torch.randn(4, requires_grad=True)
>>> p1 = pp.SO3(c)
>>> print(p1.is_leaf)
False

>>> p1.Log().sum().backward()
>>> print(c.grad)
tensor([1.0391, 0.2323, 1.4643, 0.0000])

>>> c = torch.randn(4)
>>> p1 = pp.SO3(c)
>>> print(p1.is_leaf)
True
```

wang-chen, 2023-07-27:
If you want to convert a non-leaf tensor to a leaf tensor, you can call `a.requires_grad_(True)`, where `a` can be both `Tensor` or `LieTensor`. Basically, usage of a `LieTensor` is totally the same with a `Tensor`.

zeroAska, 2023-07-27:
Thanks a lot! I have finally made it work. When I construct a least square optimizer inside `DataParallel`
 `forward()`, the pose to be optimized should take a newly-constructed `LieTensor` pose from the copy of the input `torch.Tensor` pose. It must be initialized from a *copy* of the input `torch.Tensor` instead of the original `torch.Tensor` input so that it is still a leaf.

closed 2023-07-27 by zeroAska