pypose/pypose #275: Differentiable way of construct a SE3 from SO3 and translation

zeroAska, 2023-08-24:
### 📚 The doc issue

What is the properly differentiable way to construct an SE3 object from an SO3 and 3-dim tensor?

For example:
```
R = pp.rand_SO3()
b = torch.randn((1,3))
```
Can we construct a SE3 like:
```
T = pp.SE3(R, b) 
```
such that gradient with respect to `T` can be propagated to `R` and `b`?

### Suggest a potential alternative/fix

Documentations or a new constructor of SE3

wang-chen, 2023-08-25:
Currently, we can do
```python
>>> t = torch.randn((3,), requires_grad=True)
>>> r = pp.randn_SO3(requires_grad=True)
>>> T = pp.SE3(torch.cat((t, r.tensor()), dim=-1))
>>> T.sum().backward()
>>> t.grad
tensor([2., 2., 2.])
>>> r.grad
tensor([1., 1., 1., 1.])
```
Here `.tensor()` is to bypass a User warning, but it can still run without calling it.
We can discuss your expected use case of `pp.SE3(R, b)`, but we cannot promise right now.

zeroAska, 2023-08-25:
Thanks, I would use this way for now.

closed 2023-12-11 by zitongzhan