pypose/pypose #289: Tracing a function with pypose input fails.

NicolayP, 2023-10-13:
### 🐛 Describe the bug

When I try to trace a module that has pypose elements as input it fails. However, if I convert the input to pypose elements withing the module it works as intended. Here is a small reproducible example. 

```python
import torch
import pypose as pp

from torch.utils.tensorboard import SummaryWriter

class simpleOperation(torch.nn.Module):
    def __init__(self):
        super(simpleOperation, self).__init__()

    def forward(self, x, y):
        return x + y

class wrapperPypose(torch.nn.Module):
    def __init__(self):
        super(wrapperPypose, self).__init__()
    
    def forward(self, x, y):
        pp_x = pp.SE3(x)
        pp_y = pp.SE3(y)
        return pp_x + pp_y

def fails():
    writer = SummaryWriter("foo")
    simplePP = simpleOperation()

    x = pp.randn_SE3(1, 1)
    y = pp.randn_SE3(1, 1)

    writer.add_graph(simplePP, [x, y])

def succes():
    writer = SummaryWriter("foo")
    wrapper = wrapperPypose()

    x = torch.zeros(1, 1, 7)
    y = torch.zeros(1, 1, 7)

    x[..., -1] = 1.
    y[..., -1] = 1.

    writer.add_graph(wrapper, [x, y])

if __name__ == "__main__":
    succes()
```

It's inconvenient as I need to change the internal representation of my data in my project. Going from pypose.SE3->tensor->pypose.SE3 

From the Traceback it seems like some values use are not set.

```bash
torch.jit._trace.TracingCheckError: Tracing failed sanity checks!
encountered an exception while running the trace with test inputs.
Exception:
        'torch._C.ScriptMethod' object has no attribute '__name__'   
```

### Versions

```bash
Collecting environment information...
PyTorch version: 2.0.0+cu117
Is debug build: False
CUDA used to build PyTorch: 11.7
ROCM used to build PyTorch: N/A

OS: Ubuntu 18.04.6 LTS (x86_64)
GCC version: (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0
Clang version: Could not collect
CMake version: version 3.26.3
Libc version: glibc-2.27

Python version: 3.9.16 (main, Mar  8 2023, 14:00:05)  [GCC 11.2.0] (64-bit runtime)
Python platform: Linux-5.4.0-150-generic-x86_64-with-glibc2.27
Is CUDA available: True
CUDA runtime version: Could not collect
GPU models and configuration: GPU 0: NVIDIA GeForce GTX 1650
Nvidia driver version: 525.105.17
cuDNN version: Probably one of the following:
/usr/lib/x86_64-linux-gnu/libcudnn.so.8.9.5
/usr/lib/x86_64-linux-gnu/libcudnn_adv_infer.so.8.9.5
/usr/lib/x86_64-linux-gnu/libcudnn_adv_train.so.8.9.5
/usr/lib/x86_64-linux-gnu/libcudnn_cnn_infer.so.8.9.5
/usr/lib/x86_64-linux-gnu/libcudnn_cnn_train.so.8.9.5
/usr/lib/x86_64-linux-gnu/libcudnn_ops_infer.so.8.9.5
/usr/lib/x86_64-linux-gnu/libcudnn_ops_train.so.8.9.5
/usr/local/cuda-11.7/targets/x86_64-linux/lib/libcudnn.so.8
/usr/local/cuda-11.7/targets/x86_64-linux/lib/libcudnn_adv_infer.so.8
/usr/local/cuda-11.7/targets/x86_64-linux/lib/libcudnn_adv_train.so.8
/usr/local/cuda-11.7/targets/x86_64-linux/lib/libcudnn_cnn_infer.so.8
/usr/local/cuda-11.7/targets/x86_64-linux/lib/libcudnn_cnn_train.so.8
/usr/local/cuda-11.7/targets/x86_64-linux/lib/libcudnn_ops_infer.so.8
/usr/local/cuda-11.7/targets/x86_64-linux/lib/libcudnn_ops_train.so.8
HIP runtime version: N/A
MIOpen runtime version: N/A
Is XNNPACK available: True

Versions of relevant libraries:
[pip3] numpy==1.24.3
[pip3] torch==2.0.0
[pip3] torchvision==0.15.1
[conda] numpy                     1.24.3                   pypi_0    pypi
[conda] torch                     2.0.0                    pypi_0    pypi
[conda] torchvision               0.15.1                   pypi_0    pypi
```
the version of pypose running is 0.6.3

wang-chen, 2023-10-13:
@zitongzhan, this bug may also exist in SbkTensor, please check, I am trying to solve it with #290.

wang-chen, 2023-10-13:
@NicolayP, a new version `v0.6.4` has been released to fix this bug, could you try it?
You may need `pip install -U pypose`

NicolayP, 2023-10-16:
Yep that seems to be working. Thanks for your help! :)

closed 2023-10-16 by NicolayP