pypose/pypose #328: Error: Use reshape instead of view

haarisr, 2024-01-04:
https://github.com/pypose/pypose/blob/8928c7fd6c56fd2f0e65e19be38ebb65feba08ed/pypose/optim/optimizer.py#L38

I get a runtime error at the above line. 

```
RuntimeError: view size is not compatible with input tensor's size and stride (at least one dimension spans across two contiguous subspaces). Use .reshape(...) instead.
```

I tried using reshape and it works as expected.

I believe this has to do with the jacobian computed by pytorch and not my tensors? If this is the case I can make a PR to replace view with reshape.

wang-chen, 2024-01-04:
Hi @haarisr, yes, it sounds reasonable, please submit a PR. thank you.

wang-chen, 2024-01-13:
@haarisr your changes have been released in [v0.6.7](https://pypi.org/project/pypose/).

PR #329 (2024-01-04): fix: use reshape instead of view
closes #328