pypose/pypose #330: LM deteriorated accuracy between versions 0.6.5 -> 0.6.6

gmedan, 2024-01-11:
### 🐛 Describe the bug

I experienced a severe deterioration in accuracy and number of iterations solving a small bundle adjustment problem with LM optimizer when moving from v0.6.5 to v0.6.6.

I wasn't able to pinpoint the cause and generate a minimal reproducer, but hopefully the [code](https://gist.github.com/gmedan/0369ef61e544956886568b3989be5138) itself is short enough to look into.
The data is synthetic and is generated by the same forward method using ground-truth values, which are withheld from the solver.
I ran the exact same code on the exact same environment, only difference was pypose version.

v0.6.5 output: the problem is solved to numerical precision
```
StopOnPlateau on step 0 Loss 1.910105e+00 --> Loss 1.312729e+00 (reduction/loss: 3.1274e-01).
StopOnPlateau on step 1 Loss 1.312729e+00 --> Loss 3.240422e-01 (reduction/loss: 7.5315e-01).
StopOnPlateau on step 2 Loss 3.240422e-01 --> Loss 4.624002e-02 (reduction/loss: 8.5730e-01).
StopOnPlateau on step 3 Loss 4.624002e-02 --> Loss 2.900029e-03 (reduction/loss: 9.3728e-01).
StopOnPlateau on step 4 Loss 2.900029e-03 --> Loss 2.628294e-05 (reduction/loss: 9.9094e-01).
StopOnPlateau on step 5 Loss 2.628294e-05 --> Loss 3.116393e-09 (reduction/loss: 9.9988e-01).
StopOnPlateau on step 6 Loss 3.116393e-09 --> Loss 1.382084e-16 (reduction/loss: 1.0000e+00).
StopOnPlateau on step 7 Loss 1.382084e-16 --> Loss 1.942663e-24 (reduction/loss: 1.0000e+00).
StopOnPlateau on step 8 Loss 1.942663e-24 --> Loss 5.115759e-31 (reduction/loss: 1.0000e+00).
StopOnPlateau on step 9 Loss 5.115759e-31 --> Loss 4.304341e-31 (reduction/loss: 1.3268e-01).
StopOnPlateau: Maximum patience steps reached, Quiting..
Err Rot (deg):  0.000000000 | Err Trans (m):  0.000000000
```

v0.6.6 output: the optimizer makes different, suboptimal steps, and gives up eventually
```
StopOnPlateau on step 0 Loss 1.910105e+00 --> Loss 1.312729e+00 (reduction/loss: 3.1274e-01).
StopOnPlateau on step 1 Loss 1.312729e+00 --> Loss 7.613335e-01 (reduction/loss: 4.2004e-01).
StopOnPlateau on step 2 Loss 7.613335e-01 --> Loss 5.804327e-01 (reduction/loss: 2.3761e-01).
StopOnPlateau on step 3 Loss 5.804327e-01 --> Loss 3.535811e-01 (reduction/loss: 3.9083e-01).
StopOnPlateau on step 4 Loss 3.535811e-01 --> Loss 2.337226e-01 (reduction/loss: 3.3898e-01).
StopOnPlateau on step 5 Loss 2.337226e-01 --> Loss 1.754215e-01 (reduction/loss: 2.4945e-01).
StopOnPlateau on step 6 Loss 1.754215e-01 --> Loss 1.431928e-01 (reduction/loss: 1.8372e-01).
StopOnPlateau on step 7 Loss 1.431928e-01 --> Loss 1.145350e-01 (reduction/loss: 2.0013e-01).
StopOnPlateau on step 8 Loss 1.145350e-01 --> Loss 8.521404e-02 (reduction/loss: 2.5600e-01).
StopOnPlateau on step 9 Loss 8.521404e-02 --> Loss 5.671785e-02 (reduction/loss: 3.3441e-01).
StopOnPlateau on step 10 Loss 5.671785e-02 --> Loss 3.159629e-02 (reduction/loss: 4.4292e-01).
StopOnPlateau on step 11 Loss 3.159629e-02 --> Loss 1.359741e-02 (reduction/loss: 5.6965e-01).
StopOnPlateau on step 12 Loss 1.359741e-02 --> Loss 5.365757e-03 (reduction/loss: 6.0538e-01).
StopOnPlateau on step 13 Loss 5.365757e-03 --> Loss 3.296073e-03 (reduction/loss: 3.8572e-01).
StopOnPlateau on step 14 Loss 3.296073e-03 --> Loss 2.973364e-03 (reduction/loss: 9.7907e-02).
StopOnPlateau on step 15 Loss 2.973364e-03 --> Loss 2.541876e-03 (reduction/loss: 1.4512e-01).
StopOnPlateau on step 16 Loss 2.541876e-03 --> Loss 1.769571e-03 (reduction/loss: 3.0383e-01).
StopOnPlateau on step 17 Loss 1.769571e-03 --> Loss 9.846931e-04 (reduction/loss: 4.4354e-01).
StopOnPlateau on step 18 Loss 9.846931e-04 --> Loss 4.406246e-04 (reduction/loss: 5.5253e-01).
StopOnPlateau on step 19 Loss 4.406246e-04 --> Loss 1.681205e-04 (reduction/loss: 6.1845e-01).
StopOnPlateau on step 20 Loss 1.681205e-04 --> Loss 7.312793e-05 (reduction/loss: 5.6503e-01).
StopOnPlateau on step 21 Loss 7.312793e-05 --> Loss 5.315893e-05 (reduction/loss: 2.7307e-01).
StopOnPlateau on step 22 Loss 5.315893e-05 --> Loss 4.814676e-05 (reduction/loss: 9.4286e-02).
StopOnPlateau on step 23 Loss 4.814676e-05 --> Loss 3.847422e-05 (reduction/loss: 2.0090e-01).
StopOnPlateau on step 24 Loss 3.847422e-05 --> Loss 2.506154e-05 (reduction/loss: 3.4861e-01).
StopOnPlateau: Maximum steps reached, Quiting..
Err Rot (deg):  0.014075563 | Err Trans (m):  0.000058479
```
More steps lead to stopping on patience with large error, not to convergence.

### Versions

```
PyTorch version: 2.1.2+cu121
Is debug build: False
CUDA used to build PyTorch: 12.1
ROCM used to build PyTorch: N/A

OS: Ubuntu 20.04.6 LTS (x86_64)
GCC version: (Ubuntu 9.4.0-1ubuntu1~20.04.2) 9.4.0
Clang version: Could not collect
CMake version: version 3.16.3
Libc version: glibc-2.31

Python version: 3.10.13 (main, Aug 25 2023, 13:20:03) [GCC 9.4.0] (64-bit runtime)
Python platform: Linux-6.2.0-39-generic-x86_64-with-glibc2.31
Is CUDA available: True
CUDA runtime version: Could not collect
GPU models and configuration: GPU 0: NVIDIA GeForce RTX 3090
Nvidia driver version: 525.147.05
cuDNN version: Probably one of the following:
/usr/lib/x86_64-linux-gnu/libcudnn.so.8.8.0
/usr/lib/x86_64-linux-gnu/libcudnn_adv_infer.so.8.8.0
/usr/lib/x86_64-linux-gnu/libcudnn_adv_train.so.8.8.0
/usr/lib/x86_64-linux-gnu/libcudnn_cnn_infer.so.8.8.0
/usr/lib/x86_64-linux-gnu/libcudnn_cnn_train.so.8.8.0
/usr/lib/x86_64-linux-gnu/libcudnn_ops_infer.so.8.8.0
/usr/lib/x86_64-linux-gnu/libcudnn_ops_train.so.8.8.0
HIP runtime version: N/A
MIOpen runtime version: N/A
Is XNNPACK available: True

Versions of relevant libraries:
[pip3] mypy-extensions==1.0.0
[pip3] numpy==1.26.2
[pip3] pytorch-ignite==0.4.11
[pip3] pytorch3d==0.7.5
[pip3] torch==2.1.2
[pip3] torchvision==0.16.2
[conda] Could not collect
```

wang-chen, 2024-01-11:
Hi, @zitongzhan, It might be because of the new backward functions, can you have a look?

closed 2024-01-13 by wang-chen

wang-chen, 2024-01-13:
@gmedan [v0.6.7](https://pypi.org/project/pypose/) should have fixed your bug report. You may try `pip install -U pypose`

commit 218cd85a (2024-01-13): fix #330 (#331)

* fix #330 

* fix rest of the `ctx.saved_tensors[0]`

* amend

---------