pypose/pypose #337: Lie tensor conversion

VladimirYugay, 2024-02-21:
Hey there, I run the following test:

```
twist_a = torch.tensor([1, 2, 3, 4, 5, 6]).float()
twist_b = torch.tensor([1, 2, 4, 4, 5, 6]).float()
twist_c = torch.tensor([1, 2, 3, 4, 5, 8]).float()
```

the difference between twist a and b is the third coordinate in the "rotation" part of the twist, and the difference between twist a and c is in the 6th coordinate of the "translation" part of the twist. 

However, when I call:
```
pp.LieTensor(twist_a, ltype=pp.se3_type).matrix()
pp.LieTensor(twist_b, ltype=pp.se3_type).matrix()
```
the difference between matrices is not only in rotation but also in translation. 

When I call:
```
pp.LieTensor(twist_a, ltype=pp.se3_type).matrix()
pp.LieTensor(twist_c, ltype=pp.se3_type).matrix()
```
the difference between matrices is not only in translation but also in rotation.

I was wondering, why when I convert a twist to a transformation matrix, the rotation and translation components are not independent.

wang-chen, 2024-02-21:
Hi, you called `se3` not `SE3`. You can refer to [Log](https://pypose.org/docs/main/generated/pypose.Log/) and [Exp](https://pypose.org/docs/main/generated/pypose.Exp/) to see the relation between Lie Algebra and Lie Group. I guess you want Lie Group (`SE3`).

If you want to covert Euler angles to SO3, you may need [euler2SO3](https://pypose.org/docs/main/generated/pypose.euler2SO3/).

The documentation for quaternions is [here](https://pypose.org/docs/main/generated/pypose.SO3/#pypose.SO3).

Here is a sample code:

```
import torch, pypose as pp

euler = torch.tensor([4, 5, 6], dtype=torch.float32)
SO3 = pp.euler2SO3(euler)
trans = torch.tensor([1, 2, 3], dtype=torch.float32)
SE3 = pp.SE3(torch.cat([trans, SO3]))
print(SE3)
```
```
SE3Type LieTensor:
LieTensor([ 1.0000,  2.0000,  3.0000,  0.7563,  0.1438,  0.5858, -0.2533])
```

wang-chen, 2024-02-21:
The short answer is that the rotation and translation are independent in the representation of Lie Group, and look not independent in Lie Algebra because their relationship is (y is Lie Group)
[image]

closed 2025-01-11 by wang-chen