pypose/pypose #347: 2D Case

SuperN1ck, 2024-06-27:
### 🚀 The feature, motivation and pitch

Hey `pypose`-maintainers,
I used `pypose`'s 3D interface for a few projects, thank you so much for this awesome library! 

In one of my recent projects I wanted to test my ideas on a simpler setting and thus, decided to implement a 2D version of the problem using `pypose` to represent `SE(2)`-poses. Unfortunately though, I realized `pypose` did not explicitly include the 2D case. As I was used to the `pypose` ecosystem, I decided to copy and adapt a few of the current 3D files to 2D, e.g. implementing a `SO2_type`, etc. I wanted to ask if there is interest in such an extension of the library. If yes, I am happy to start a PR and dump mt current state and then work together on a fully integrable solution.

Looking forward to your thoughts!

### Alternatives

For working in 2D, one could obviously still use the full 3D interface and implicitly reduce the dimensions. Nonetheless, I think a native 2D solution is beneficial.

### Additional context

Previously I worked quite extensively with [jaxlie](https://github.com/brentyi/jaxlie) which offered an interface to 2D.

wang-chen, 2024-06-27:
Hi, @SuperN1ck, yes, a 2D case is definitely one of our plans and it would be great if you could create a PR.  It would also be great if your PR could follow the current code structure. Thank you so much!

shim94kr, 2024-07-29:
@SuperN1ck Hello,

I'm very interested in your implementation of the 2D version. I was wondering if there have been any recent developments or if you have an estimated timeline for its completion. I understand you must be busy, and I appreciate any information you can share!

SuperN1ck, 2024-07-29:
Hey @shim94kr,
there is this PR/branch: https://github.com/SuperN1ck/pypose/tree/liegroups_2D, from which you could pull that gives you already basic implementation for simple 2D functions. 
I unfortunately have no fixed timeline and can tell you when I will add more things. I am currently preparing the random sampling. But if you have any specific things in mind that you need soon, let me know. 
Nonetheless, also feel free to contribute, you are more than welcome. 
Cheers,
-Nick!

shim94kr, 2024-07-30:
Hey Nick,

I wasn't aware there was already an implemented function. I'll make good use of it, thanks for letting me know. Best of luck with the upcoming implementations! It sounds like meaningful work.

Take care!