pypose/pypose #348: Pose Graph Optimization: Unstable Results in pgo_tutorial.py

gogojjh, 2024-06-27:
### 🐛 Describe the bug

I created a simple g2o file to test the pose graph optimization. Here is the g2o: 
```
VERTEX_SE3:QUAT 0 0 0.1 0 0 0 0 1
VERTEX_SE3:QUAT 1 2.3 0.1 0 -0.0998334 0 0 0.995004
VERTEX_SE3:QUAT 2 4.1 0.1 0 0 0 0.707107 0.707107
VERTEX_SE3:QUAT 3 4 2 0 0 0 1 6.12323e-17
VERTEX_SE3:QUAT 4 2.1 2.1 0 -0 0 1 -6.12323e-17
EDGE_SE3:QUAT 0 1 2 0 0 0 0 0 1 25 0 0 0 0 0 25 0 0 0 0 25 0 0 0 25 0 0 25 0 25
EDGE_SE3:QUAT 1 2 2 0 0 0 0 0.707107 0.707107 25 0 0 0 0 0 25 0 0 0 0 25 0 0 0 25 0 0 25 0 25
EDGE_SE3:QUAT 2 3 2 0 0 0 0 0.707107 0.707107 25 0 0 0 0 0 25 0 0 0 0 25 0 0 0 25 0 0 25 0 25
EDGE_SE3:QUAT 3 4 2 0 0 0 0 0.707107 0.707107 25 0 0 0 0 0 25 0 0 0 0 25 0 0 0 25 0 0 25 0 25
EDGE_SE3:QUAT 4 1 2 0 0 0 0 0.707107 0.707107 25 0 0 0 0 0 25 0 0 0 0 25 0 0 0 25 0 0 25 0 25
```
I have verified the results with the GTSAM[Pose3SLAMExample_g2o.py](https://github.com/borglab/gtsam/blob/develop/python/gtsam/examples/Pose3SLAMExample_g2o.py).

I ran the code from the following tutorial: [pgo_tutorial.py](https://github.com/pypose/tutorials/blob/main/pgo_source/pgo_tutorial.py) and observed that the results are unstable. To correct this, we should impose a prior factor on the first pose and covariance information.
I would like to understand the proper method to impose prior constraints. While I am reporting this issue here, I am unsure if it qualifies as a bug. 
Thank you for your assistance.

### Versions

Collecting environment information...
PyTorch version: 2.0.1+cu117
Is debug build: False
CUDA used to build PyTorch: 11.7
ROCM used to build PyTorch: N/A

OS: Ubuntu 20.04.6 LTS (x86_64)
GCC version: (Ubuntu 9.4.0-1ubuntu1~20.04.2) 9.4.0
Clang version: 10.0.0-4ubuntu1 
CMake version: version 3.29.6
Libc version: glibc-2.31

Python version: 3.8.10 (default, Nov 22 2023, 10:22:35)  [GCC 9.4.0] (64-bit runtime)
Python platform: Linux-5.15.0-101-generic-x86_64-with-glibc2.29
Is CUDA available: True
CUDA runtime version: 11.8.89
GPU models and configuration: GPU 0: NVIDIA GeForce RTX 4090
Nvidia driver version: 535.161.07
cuDNN version: Could not collect
HIP runtime version: N/A
MIOpen runtime version: N/A
Is XNNPACK available: True

Versions of relevant libraries:
[pip3] numpy==1.23.5
[pip3] rosnumpy==0.0.5.2
[pip3] smplpytorch==0.0.8
[pip3] torch==2.0.1
[conda] No relevant packages

wang-chen, 2025-01-11:
To add a prior constraint, an easy way is to concatenate new error terms in the forward function like "self.nodes[0].Inv() * init_pose".

wang-chen, 2025-01-11:
It also allows [covariance matrices (weights)](https://pypose.org/docs/main/generated/pypose.optim.LevenbergMarquardt/#pypose.optim.LevenbergMarquardt.step) in the step function. If covariance matrices are pre-defined and fixed, you may also set it when you define an [optimizer](https://pypose.org/docs/main/generated/pypose.optim.LevenbergMarquardt/#pypose.optim.LevenbergMarquardt) using the `weight` argument.

closed 2025-01-11 by wang-chen

gogojjh, 2025-01-12:
Thanks for your reply!