pypose/pypose #354: Tracing issues

ignotuspeverel, 2024-12-02:
### 📚 The doc issue

Hi, I'm now currently using pypose to track in SLAM. I'm now using a gradient-based method, and the system can work with a hand-written se3pose [https://github.com/zju3dv/Vox-Fusion/blob/main/src/se3pose.py]. However when using pypose, I cannot find out a good parameter for the system to converge and it will always fall into local minimum. Can you give any suggestions, many thanks.

wang-chen, 2024-12-02:
hi, gradient-based methods basically cannot work for optimizing poses, there are many reasons behind this. Your implementation is updating matrices, which may result in invalid pose or rotation matrices (not orthogonal anymore) after multiple iterations. You'll need 2nd-order optimization for optiming poses.

ignotuspeverel, 2024-12-03:
So using methods like LM optimization can work for optimizing the poses ? I will give it a try.

closed 2025-01-11 by wang-chen