pypose/pypose #355: Pose Graph Optimization with GPS and Loop Closure Constraints - Different Results on GPU and CPU

RomanStadlhuber, 2024-12-05:
I am sorry somehow I could not submit through the bug report template.

#### 🐛 Describe the bug

I want to optimize a Visual Odometry trajectory using GPS data and loop closure based on GPS locations.
The VO trajectory comes pre-scaled and aligned to initialize the PGO and is converted to SE(3) since the initial scaling works quite well already.
VO poses are synced with GPS where possible and loop closure edges and constraints from GPS are also passed to the pose graph.

When I run the below graph on `device="cuda`, the results are quite nice, but when `device="cpu"`, the poses that could not be associated with a GPS constraint are very incorrect.
I want to inquire as to where this behavior comes from and if there is anything that can be done about it.
Or maybe there is some other thing that I am just not seeing.

When running on CUDA:
[image]

and when running the same PGO on CPU:
[image]

#### Code and Data

<details>
<summary>import dependencies</summary>

```python
# PGO
import torch
from torch import nn
import pypose as pp
import pypose.optim.solver as ppos
import pypose.optim.strategy as ppost
from pypose.optim.scheduler import StopOnPlateau
# to load data
import pickle
import numpy as np
```
</details>

You can [find the data here](https://drive.google.com/drive/folders/1ADkBuCFK6aYOltz8rXF4wk1inqzRXPBC?usp=sharing) and load it like below

```python
poses_initial = pp.SE3(np.load("poses_initial.npy"))
slam_edges = np.load("slam_edges.npy")
lc_edges = np.load("lc_edges.npy")
lc_errors = np.load("lc_errors.npy")
with open("gps.pkl", "rb") as f:
    gps = pickle.load(f)
```
force usage of CPU

```python
# device = "cuda" if torch.cuda.is_available() else "cpu"
device = "cpu"
print(f"optimizer will run on {device}")
```

Then define the graph to be optimized

```python
class PoseGraph(nn.Module):
    def __init__(
        self,
        poses: pp.LieTensor,
        gps: list[np.ndarray | None],
        loop_edges: np.ndarray,
        loop_distances: np.ndarray,
    ) -> None:
        super().__init__()
        self.poses = pp.Parameter(poses)
        self.gps = gps
        self._num_constraints = len(gps)
        self.loop_edges = loop_edges
        self.loop_distances = torch.tensor(loop_distances.astype(np.float32)).to(device)
        self.num_loops, *_ = self.loop_edges.shape
        ## initialize residuals that will be used throughout the network operation ###
        self.gps_residual = torch.zeros(3, self._num_constraints).to(device)
        self.gps_pts = torch.tensor(np.array([pt if pt is not None else np.zeros((3,)) for pt in self.gps])).to(device)
        self.gps_idxs = [en[0] for en in enumerate(self.gps) if en[1] is not None]

    def forward(self, edges, poses):
        poses.to(device)
        X1 = self.poses[edges[..., 0]]
        X2 = self.poses[edges[..., 1]]
        X1_init = poses[edges[..., 0]].to(device)
        X2_init = poses[edges[..., 1]].to(device)
        # relative pose error
        # NOTE: the first pose cannot be optimized relative to anything, it is the origin
        pose_delta = X1.Inv() @ X2
        error = (X1_init.Inv() @ X2_init).Inv() @ pose_delta
        pose_residual = torch.transpose(error.Log().tensor(), 0, 1)
        # GPS residual stays zero if there is pose does not contain a sync entry
        for idx in self.gps_idxs:
            self.gps_residual[:, idx] = self.poses[idx, :3] - self.gps_pts[idx, :]
        total_error = torch.cat((pose_residual, self.gps_residual)).to(device)
        if self.num_loops > 0:
            # dim(err) x N_lc
            loop_residual = torch.zeros((total_error.shape[0], self.num_loops)).to(device)
            X_lc_source = self.poses[self.loop_edges[..., 0]]
            X_lc_target = self.poses[self.loop_edges[..., 1]]
            # N_lc x 3
            loop_error = (X_lc_target.Inv() @ X_lc_source).Log()[..., :3]
            # transpose to 3 x N_lc
            loop_residual[:3, ...] = torch.transpose(loop_error - self.loop_distances, 0, 1)
            # NOTE: loop closure error is position based
            # -> append as additinal pose error!
            total_error = torch.cat((total_error, loop_residual), dim=1)
        # NOTE: residual always needs to be (error_dim, N)!
        # see: https://github.com/pypose/pypose/issues/256#issuecomment-1633490219
        return total_error
```
lastly, setup and run the scheduler

```python
graph = PoseGraph(poses_initial, gps, lc_edges, lc_errors).to(device)
solver = ppos.Cholesky()
strategy = ppost.TrustRegion(radius=1e-2)
optimizer = pp.optim.LM(graph, solver=solver, strategy=strategy, min=1e-6)
scheduler = StopOnPlateau(optimizer, steps=60, patience=10, decreasing=1e-7, verbose=True)
scheduler.optimize(input=(slam_edges, poses_initial))
```

It will converge in both cases, but take a different amount of steps and produce different results like above.

This issue might not be as devastating in the presented case, but when there are fewer GPS syncs (e.g. when passing through a GPS-denied region) this causes all of these poses to be incorrectly optimized.

#### Versions

```
$ python collect_env.py 
Collecting environment information...
PyTorch version: 2.5.1+cu124
Is debug build: False
CUDA used to build PyTorch: 12.4
ROCM used to build PyTorch: N/A

OS: Ubuntu 22.04.5 LTS (x86_64)
GCC version: (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0
Clang version: Could not collect
CMake version: version 3.22.1
Libc version: glibc-2.35

Python version: 3.10.12 (main, Nov  6 2024, 20:22:13) [GCC 11.4.0] (64-bit runtime)
Python platform: Linux-6.5.0-1027-oem-x86_64-with-glibc2.35
Is CUDA available: True
CUDA runtime version: 11.5.119
GPU models and configuration: GPU 0: NVIDIA RTX A500 Laptop GPU
Nvidia driver version: 535.183.01
cuDNN version: Could not collect
HIP runtime version: N/A
MIOpen runtime version: N/A

Is XNNPACK available: True

Versions of relevant libraries:
[pip3] numpy==1.26.4
[pip3] torch==2.5.1
[conda] Could not collect
```

Please let me know if there is any information you are missing.

RomanStadlhuber, 2025-01-08:
bump
any ideas if and when this might be looked at?

wang-chen, 2025-01-08:
@zitongzhan are you able to have a look? Thank you!

wang-chen, 2025-01-10:
@RomanStadlhuber Could you share a fully runnable code to reproduce your results, including drawing figures?

wang-chen, 2025-01-11:
Hi, @RomanStadlhuber Just running this code, I found both cases can optimize well, error ``3.62e+04 --> 3.32e+00``
The final optimization error using CPU is even slightly smaller than using GPU, although it is not visually looking good. 
It might be because of the local minimum in 2nd-order optimization, which often happens when dealing with "very unaligned" data measurements from different sensors. It is not a bug in the optimization functions.

RomanStadlhuber, 2025-01-11:
> It might be because of the local minimum in 2nd-order optimization, which often happens when dealing with "very unaligned" data measurements from different sensors. It is not a bug in the optimization functions.

@wang-chen but that does not explain why the results are different for CPU and CUDA + why only the poses without an additional tensor residual (GPS) are the ones that aren't optimized on CPU?

wang-chen, 2025-01-11:
hi, if that happens, it may be because you frequently move data between CPU and GPU within the ``model.forward`` and gradients are not propagated properly (it was cut at some point).

My suggestions: 
Try to prepare all data in the same devices before you define any graphs and optimizers.
Don't use `.to(device)` within a model;
Use `torch.zeros(..., device=device)`, if you have to create data in a model.

RomanStadlhuber, 2025-01-11:
thx I will try that and get back if I encounter any further problems!

wang-chen, 2025-01-11:
> thx I will try that and get back if I encounter any further problems!

I just tried the above suggestions, I can now get the same results on both CPU and GPU. Good Luck!

RomanStadlhuber, 2025-01-13:
Hello, I am sorry reopen this issue, but the problem of different behavior of the optimizer persists with different data.
Namely, when `device="cpu"`, optimization runs until the max number of steps is exhausted whereas it stops after only a few when `device="cuda"`.
In the below examples, the results from CUDA are preferred, it seems that the PGO is overfit when using the CPU!

[image]

You can find a [**runnable script** in my gist](https://gist.github.com/RomanStadlhuber/54d9a00b3f3f89a3cdfbf81aed2030fb), it downloads the data using `gdown`, so please don't forget to `pip install gdown` before running it.

To run with cuda, omit the the argument or do

```bash
python pgo.py cuda
```
<details><summary>Output</summary>

```
Device: cuda
/pgo/.venv/lib/python3.10/site-packages/pypose/lietensor/lietensor.py:918: UserWarning: Tensor Shape Invalid by calling <slot wrapper '__getitem__' of 'torch._C.TensorBase' objects>, go to https://pypose.org/docs/main/generated/pypose.LieTensor
  warnings.warn('Tensor Shape Invalid by calling {}, ' \
StopOnPlateau on step 0 Loss 5.251633e+00 --> Loss 5.251623e+00 (reduction/loss: 1.8160e-06).
StopOnPlateau on step 1 Loss 5.251623e+00 --> Loss 5.251604e+00 (reduction/loss: 3.6319e-06).
StopOnPlateau on step 2 Loss 5.251604e+00 --> Loss 5.251567e+00 (reduction/loss: 7.0823e-06).
StopOnPlateau on step 3 Loss 5.251567e+00 --> Loss 5.251494e+00 (reduction/loss: 1.3892e-05).
StopOnPlateau on step 4 Loss 5.251494e+00 --> Loss 5.251354e+00 (reduction/loss: 2.6604e-05).
StopOnPlateau on step 5 Loss 5.251354e+00 --> Loss 5.251090e+00 (reduction/loss: 5.0396e-05).
StopOnPlateau on step 6 Loss 5.251090e+00 --> Loss 5.250594e+00 (reduction/loss: 9.4349e-05).
StopOnPlateau on step 7 Loss 5.250594e+00 --> Loss 5.249683e+00 (reduction/loss: 1.7355e-04).
StopOnPlateau on step 8 Loss 5.249683e+00 --> Loss 5.248046e+00 (reduction/loss: 3.1173e-04).
StopOnPlateau on step 9 Loss 5.248046e+00 --> Loss 5.245169e+00 (reduction/loss: 5.4834e-04).
StopOnPlateau on step 10 Loss 5.245169e+00 --> Loss 5.240169e+00 (reduction/loss: 9.5328e-04).
StopOnPlateau on step 11 Loss 5.240169e+00 --> Loss 5.231513e+00 (reduction/loss: 1.6518e-03).
StopOnPlateau on step 12 Loss 5.231513e+00 --> Loss 5.216715e+00 (reduction/loss: 2.8287e-03).
StopOnPlateau on step 13 Loss 5.216715e+00 --> Loss 5.192423e+00 (reduction/loss: 4.6566e-03).
StopOnPlateau on step 14 Loss 5.192423e+00 --> Loss 5.155973e+00 (reduction/loss: 7.0197e-03).
StopOnPlateau on step 15 Loss 5.155973e+00 --> Loss 5.126920e+00 (reduction/loss: 5.6350e-03).
StopOnPlateau on step 16 Loss 5.126920e+00 --> Loss 5.104273e+00 (reduction/loss: 4.4173e-03).
StopOnPlateau on step 17 Loss 5.104273e+00 --> Loss 5.087349e+00 (reduction/loss: 3.3156e-03).
StopOnPlateau on step 18 Loss 5.087349e+00 --> Loss 5.075520e+00 (reduction/loss: 2.3253e-03).
StopOnPlateau on step 19 Loss 5.075520e+00 --> Loss 5.068123e+00 (reduction/loss: 1.4573e-03).
StopOnPlateau on step 20 Loss 5.068123e+00 --> Loss 5.064476e+00 (reduction/loss: 7.1966e-04).
StopOnPlateau on step 21 Loss 5.064476e+00 --> Loss 5.063922e+00 (reduction/loss: 1.0931e-04).
StopOnPlateau on step 22 Loss 5.063922e+00 --> Loss 5.063561e+00 (reduction/loss: 7.1188e-05).
StopOnPlateau: Maximum rejected steps reached, Quiting..

```

</details>

to run on CPU, do

```bash
python pgo.py cpu
```

<details><summary>Output</summary>

```
Device: cpu
/pgo/.venv/lib/python3.10/site-packages/pypose/lietensor/lietensor.py:918: UserWarning: Tensor Shape Invalid by calling <slot wrapper '__getitem__' of 'torch._C.TensorBase' objects>, go to https://pypose.org/docs/main/generated/pypose.LieTensor
  warnings.warn('Tensor Shape Invalid by calling {}, ' \
StopOnPlateau on step 0 Loss 5.251633e+00 --> Loss 5.251623e+00 (reduction/loss: 1.8160e-06).
StopOnPlateau on step 1 Loss 5.251623e+00 --> Loss 5.251604e+00 (reduction/loss: 3.6319e-06).
StopOnPlateau on step 2 Loss 5.251604e+00 --> Loss 5.251566e+00 (reduction/loss: 7.2639e-06).
StopOnPlateau on step 3 Loss 5.251566e+00 --> Loss 5.251491e+00 (reduction/loss: 1.4255e-05).
StopOnPlateau on step 4 Loss 5.251491e+00 --> Loss 5.251344e+00 (reduction/loss: 2.7967e-05).
StopOnPlateau on step 5 Loss 5.251344e+00 --> Loss 5.251059e+00 (reduction/loss: 5.4391e-05).
StopOnPlateau on step 6 Loss 5.251059e+00 --> Loss 5.250509e+00 (reduction/loss: 1.0461e-04).
StopOnPlateau on step 7 Loss 5.250509e+00 --> Loss 5.249466e+00 (reduction/loss: 1.9871e-04).
StopOnPlateau on step 8 Loss 5.249466e+00 --> Loss 5.247514e+00 (reduction/loss: 3.7179e-04).
StopOnPlateau on step 9 Loss 5.247514e+00 --> Loss 5.243923e+00 (reduction/loss: 6.8434e-04).
[... 71 lines cut ...]
StopOnPlateau on step 81 Loss 3.544860e+00 --> Loss 3.538529e+00 (reduction/loss: 1.7861e-03).
StopOnPlateau on step 82 Loss 3.538529e+00 --> Loss 3.532294e+00 (reduction/loss: 1.7619e-03).
StopOnPlateau on step 83 Loss 3.532294e+00 --> Loss 3.526153e+00 (reduction/loss: 1.7387e-03).
StopOnPlateau on step 84 Loss 3.526153e+00 --> Loss 3.520098e+00 (reduction/loss: 1.7171e-03).
StopOnPlateau on step 85 Loss 3.520098e+00 --> Loss 3.514126e+00 (reduction/loss: 1.6966e-03).
StopOnPlateau on step 86 Loss 3.514126e+00 --> Loss 3.508231e+00 (reduction/loss: 1.6773e-03).
StopOnPlateau on step 87 Loss 3.508231e+00 --> Loss 3.502410e+00 (reduction/loss: 1.6592e-03).
StopOnPlateau on step 88 Loss 3.502410e+00 --> Loss 3.496659e+00 (reduction/loss: 1.6422e-03).
[... cut ...]
StopOnPlateau on step 96 Loss 3.458022e+00 --> Loss 3.452695e+00 (reduction/loss: 1.5405e-03).
StopOnPlateau on step 97 Loss 3.452695e+00 --> Loss 3.447406e+00 (reduction/loss: 1.5318e-03).
StopOnPlateau on step 98 Loss 3.447406e+00 --> Loss 3.442155e+00 (reduction/loss: 1.5233e-03).
StopOnPlateau on step 99 Loss 3.442155e+00 --> Loss 3.436938e+00 (reduction/loss: 1.5156e-03).
StopOnPlateau: Maximum steps reached, Quiting..

```
</details>

My versions info is still the same as above, please let me know if you need any additional information.

wang-chen, 2025-01-13:
here is your codes:

```
# PGO
import torch
from torch import nn
import pypose as pp
import pypose.optim.solver as ppos
import pypose.optim.strategy as ppost
from pypose.optim.scheduler import StopOnPlateau
from pathlib import Path
import matplotlib.pyplot as plt
import sys

# to load data
import pickle
import numpy as np
import gdown

class PoseGraph(nn.Module):
    def __init__(
        self,
        poses: pp.LieTensor,
        gps: list[np.ndarray | None],
    ) -> None:
        super().__init__()
        self.poses = pp.Parameter(poses.to(device))
        self.gps = gps
        self._num_constraints = len(gps)
        # GPS residual block will be used throughout PGO
        self.gps_residual = torch.zeros(3, self.N).to(device)
        self.gps_pts = (
            torch.tensor(np.array([pt if pt is not None else np.zeros((3,)) for pt in self.gps])).float().to(device)
        )
        self.gps_idxs = [en[0] for en in enumerate(self.gps) if en[1] is not None]

    def forward(self, edges, poses):
        X1 = self.poses[edges[..., 0]]
        X2 = self.poses[edges[..., 1]]
        X1_init = poses[edges[..., 0]]
        X2_init = poses[edges[..., 1]]
        pose_delta = X1.Inv() @ X2
        error = (X1_init.Inv() @ X2_init).Inv() @ pose_delta
        pose_residual = torch.transpose(error.Log().tensor(), 0, 1)
        # GPS residual stays zero if there is pose does not contain a sync entry
        self.gps_residual[:, self.gps_idxs] = (
            self.poses[self.gps_idxs, :3] - self.gps_pts[self.gps_idxs, :]
        ).transpose(0, 1)
        total_error = torch.cat((pose_residual, self.gps_residual))
        return total_error

    @property
    def N(self) -> int:
        """Number of model residuals."""
        return self._num_constraints

if __name__ == "__main__":
    # download from google drive, skip if it exists already
    f_gps, f_poses, f_edges = (
        gdown.download_folder(
            "https://drive.google.com/drive/folders/173wMchBIFVJjiUupsV_5YQ4j3ctwS_tk?usp=drive_link", output="data"
        )
        if not Path("data").exists()
        else ("data/gps.pkl", "data/poses.npy", "data/slam_edges.npy")
    )
    # load the data
    poses = pp.SE3(np.load(f_poses))
    slam_edges = np.load(f_edges)
    with open(f_gps, "rb") as f:
        gps = pickle.load(f)
    # device device for optimization
    device = sys.argv[1] if len(sys.argv) > 1 else ("cuda" if torch.cuda.is_available() else "cpu")
    print("Device:", device)
    # create the graph
    graph = PoseGraph(poses, gps).to(device)
    # setup the optimization process
    solver = ppos.Cholesky()
    strategy = ppost.TrustRegion(radius=1e-2)
    kernel = pp.optim.kernel.Cauchy(delta=0.5)
    optimizer = pp.optim.LM(graph, solver=solver, strategy=strategy, min=1e-6, kernel=kernel)
    scheduler = StopOnPlateau(optimizer, steps=100, patience=10, decreasing=1e-7, verbose=True)
    scheduler.optimize(input=(slam_edges, poses.to(device)))
    # show optimized results
    points_optimized = graph.poses.detach().translation().cpu().numpy()
    # points_initial = poses.translation().cpu().numpy()
    points_gps = np.array([pt if pt is not None else np.zeros((3,)) for pt in gps])
    fig = plt.figure()
    ax = fig.add_subplot(111, projection="3d")
    points_initial = np.load(f_poses)[:, :3]
    ax.plot(points_initial[:, 0], points_initial[:, 1], points_initial[:, 2], c="r", label="Initial")
    ax.plot(points_optimized[:, 0], points_optimized[:, 1], points_optimized[:, 2], c="g", label="Optimized")
    ax.plot(points_gps[:, 0], points_gps[:, 1], points_gps[:, 2], c="b", label="GPS")
    ax.set_box_aspect([1, 1, 0.1])
    ax.set_xlabel("X")
    ax.set_ylabel("Y")
    ax.set_zlabel("Z")
    ax.legend()
    plt.show()

```

RomanStadlhuber, 2025-01-14:
Ok the results are consistent now! But I also get this warning now multiple times during the optimization process.

```
/usr/local/lib/python3.10/dist-packages/pypose/lietensor/lietensor.py:919: UserWarning: Tensor Shape Invalid by calling <slot wrapper '__getitem__' of 'torch._C.TensorBase' objects>, go to https://pypose.org/docs/main/generated/pypose.LieTensor
  warnings.warn('Tensor Shape Invalid by calling {}, ' \
```

I guess it comes from indexing the pose tensor, specifically from

```python
self.poses[self.gps_idxs, :3] - self.gps_pts[self.gps_idxs, :]
```

where `type(self.poses) = pp.LieTensor`.
Any suggestions on that? I am guessing it would not impact results but it might be nice to know where this comes from.

closed 2025-01-14 by RomanStadlhuber

wang-chen, 2025-01-14:
Hi @RomanStadlhuber , 
use `self.poses[self.gps_idxs, :].translation()` or `self.poses[self.gps_idxs].translation()` 
instead of `self.poses[self.gps_idxs, :3]` 
may solve that.