pypose/pypose #360: Incorrect EKF implementation and formulation

haleqiu, 2024-12-31:
## Summary

The current EKF implementation is actually the Kalman Filter (KF) with a linear function. 

## Improvements

1.  Change the linear state propagation to nonlinear.
2.  Incorrect implementation of posteriori state update and the definition of time.
3.  Fix the covariance propagation. Current Implementation simplifies the uncertainty model of u.

wang-chen, 2025-01-11:
@haleqiu Hi, it is an EKF, since it allows a nonlinear system as the input, the system matrices [in this line](https://github.com/pypose/pypose/blob/d1e99a56b65e6db06733c9a1e77592d0078309da/pypose/module/ekf.py#L152) will automatically call the linearized system matrices [in this line](https://github.com/pypose/pypose/blob/d1e99a56b65e6db06733c9a1e77592d0078309da/pypose/module/dynamics.py#L570), if the system is an [NLS](https://pypose.org/docs/main/generated/pypose.module.NLS/?highlight=nls#pypose.module.NLS).

haleqiu, 2025-01-12:
@wang-chen I think the formulation of the NLS is correct. But the EKF formulation in `ekf.py` is incorrect.

#363

closed 2025-03-28 by wang-chen