pypose/pypose #364: Using pypose with torch.compile

lahavlipson, 2025-01-14:
### 🐛 Describe the bug

`torch.compile` doesn't seem to play nice with certain pypose functions. Example:

```
import torch
import pypose

@torch.compile
def func(a):
    return pypose.se3(a)

func(torch.zeros(1,6).cuda())
```
Traceback:

```
Traceback (most recent call last):
  File "/root/Downloads/compile_test.py", line 8, in <module>
    func(torch.zeros(1,6).cuda())
  File "/miniforge/envs/pypose/lib/python3.10/site-packages/torch/_dynamo/eval_frame.py", line 451, in _fn
    return fn(*args, **kwargs)
  File "/root/Downloads/compile_test.py", line 6, in func
    return pypose.se3(a)
  File "/miniforge/envs/pypose/lib/python3.10/site-packages/torch/_dynamo/convert_frame.py", line 921, in catch_errors
    return callback(frame, cache_entry, hooks, frame_state, skip=1)
  File "/miniforge/envs/pypose/lib/python3.10/site-packages/torch/_dynamo/convert_frame.py", line 786, in _convert_frame
    result = inner_convert(
  File "/miniforge/envs/pypose/lib/python3.10/site-packages/torch/_dynamo/convert_frame.py", line 400, in _convert_frame_assert
    return _compile(
  File "/miniforge/envs/pypose/lib/python3.10/contextlib.py", line 79, in inner
    return func(*args, **kwds)
  File "/miniforge/envs/pypose/lib/python3.10/site-packages/torch/_dynamo/convert_frame.py", line 703, in _compile
    raise InternalTorchDynamoError(str(e)).with_traceback(
  File "/miniforge/envs/pypose/lib/python3.10/site-packages/torch/_dynamo/convert_frame.py", line 676, in _compile
    guarded_code = compile_inner(code, one_graph, hooks, transform)
  File "/miniforge/envs/pypose/lib/python3.10/site-packages/torch/_dynamo/utils.py", line 262, in time_wrapper
    r = func(*args, **kwargs)
  File "/miniforge/envs/pypose/lib/python3.10/site-packages/torch/_dynamo/convert_frame.py", line 535, in compile_inner
    out_code = transform_code_object(code, transform)
  File "/miniforge/envs/pypose/lib/python3.10/site-packages/torch/_dynamo/bytecode_transformation.py", line 1036, in transform_code_object
    transformations(instructions, code_options)
  File "/miniforge/envs/pypose/lib/python3.10/site-packages/torch/_dynamo/convert_frame.py", line 165, in _fn
    return fn(*args, **kwargs)
  File "/miniforge/envs/pypose/lib/python3.10/site-packages/torch/_dynamo/convert_frame.py", line 500, in transform
    tracer.run()
  File "/miniforge/envs/pypose/lib/python3.10/site-packages/torch/_dynamo/symbolic_convert.py", line 2149, in run
    super().run()
  File "/miniforge/envs/pypose/lib/python3.10/site-packages/torch/_dynamo/symbolic_convert.py", line 810, in run
    and self.step()
  File "/miniforge/envs/pypose/lib/python3.10/site-packages/torch/_dynamo/symbolic_convert.py", line 773, in step
    getattr(self, inst.opname)(inst)
  File "/miniforge/envs/pypose/lib/python3.10/site-packages/torch/_dynamo/symbolic_convert.py", line 489, in wrapper
    return inner_fn(self, inst)
  File "/miniforge/envs/pypose/lib/python3.10/site-packages/torch/_dynamo/symbolic_convert.py", line 1219, in CALL_FUNCTION
    self.call_function(fn, args, {})
  File "/miniforge/envs/pypose/lib/python3.10/site-packages/torch/_dynamo/symbolic_convert.py", line 674, in call_function
    self.push(fn.call_function(self, args, kwargs))
  File "/miniforge/envs/pypose/lib/python3.10/site-packages/torch/_dynamo/variables/torch.py", line 681, in call_function
    proxy=tx.output.create_proxy(
  File "/miniforge/envs/pypose/lib/python3.10/site-packages/torch/_dynamo/output_graph.py", line 482, in create_proxy
    return self.current_tracer.create_proxy(*args, **kwargs)
  File "/miniforge/envs/pypose/lib/python3.10/site-packages/torch/_dynamo/output_graph.py", line 1792, in create_proxy
    rv = super().create_proxy(
  File "/miniforge/envs/pypose/lib/python3.10/site-packages/torch/fx/proxy.py", line 187, in create_proxy
    args_ = self.create_arg(args)
  File "/miniforge/envs/pypose/lib/python3.10/site-packages/torch/fx/_symbolic_trace.py", line 401, in create_arg
    return super().create_arg(a)
  File "/miniforge/envs/pypose/lib/python3.10/site-packages/torch/fx/proxy.py", line 258, in create_arg
    return type(a)(self.create_arg(elem) for elem in a)
  File "/miniforge/envs/pypose/lib/python3.10/site-packages/torch/fx/proxy.py", line 258, in <genexpr>
    return type(a)(self.create_arg(elem) for elem in a)
  File "/miniforge/envs/pypose/lib/python3.10/site-packages/torch/fx/_symbolic_trace.py", line 401, in create_arg
    return super().create_arg(a)
  File "/miniforge/envs/pypose/lib/python3.10/site-packages/torch/fx/proxy.py", line 294, in create_arg
    raise NotImplementedError(f"argument of type: {type(a)}")
torch._dynamo.exc.InternalTorchDynamoError: argument of type: <class 'torch._C._TensorMeta'>

from user code:
   File "/miniforge/envs/pypose/lib/python3.10/site-packages/pypose/lietensor/lietensor.py", line 894, in __new__
    return torch.Tensor.as_subclass(tensor, LieTensor)

Set TORCH_LOGS="+dynamo" and TORCHDYNAMO_VERBOSE=1 for more information

You can suppress this exception and fall back to eager by setting:
    import torch._dynamo
    torch._dynamo.config.suppress_errors = True
```

This issue is unrelated to #234, which I couldn't reproduce.

### Versions

I pip installed pypose from the latest commit on the main branch (sha: 44a482ac)
------------
PyTorch version: 2.3.1
Is debug build: False
CUDA used to build PyTorch: 12.1
ROCM used to build PyTorch: N/A

OS: Ubuntu 22.04.5 LTS (x86_64)
GCC version: (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0
Clang version: Could not collect
CMake version: version 3.30.3
Libc version: glibc-2.35

Python version: 3.10.16 | packaged by conda-forge | (main, Dec  5 2024, 14:16:10) [GCC 13.3.0] (64-bit runtime)
Python platform: Linux-5.10.192-183.736.amzn2.x86_64-x86_64-with-glibc2.35
Is CUDA available: True
CUDA runtime version: 12.4.131
GPU models and configuration: GPU 0: NVIDIA H100 80GB HBM3
Nvidia driver version: 535.129.03
cuDNN version: Could not collect
HIP runtime version: N/A
MIOpen runtime version: N/A
Is XNNPACK available: True

Versions of relevant libraries:
[pip3] numpy==1.26.4
[pip3] torch==2.3.1
[pip3] torch-scatter==2.1.2
[pip3] torchvision==0.18.1
[conda] blas                      1.0                         mkl    conda-forge
[conda] libblas                   3.9.0            16_linux64_mkl    conda-forge
[conda] libcblas                  3.9.0            16_linux64_mkl    conda-forge
[conda] liblapack                 3.9.0            16_linux64_mkl    conda-forge
[conda] libopenvino-pytorch-frontend 2024.6.0             h5888daf_3    conda-forge
[conda] mkl                       2022.1.0           hc2b9512_224
[conda] numpy                     1.26.4                   pypi_0    pypi
[conda] pytorch                   2.3.1           py3.10_cuda12.1_cudnn8.9.2_0    pytorch
[conda] pytorch-cuda              12.1                 ha16c6d3_6    pytorch
[conda] pytorch-mutex             1.0                        cuda    pytorch
[conda] pytorch-scatter           2.1.2           py310_torch_2.3.0_cu121    pyg
[conda] torchtriton               2.3.1                     py310    pytorch
[conda] torchvision               0.18.1              py310_cu121    pytorch

zitongzhan, 2025-01-16:
Looks like the error message shows that tensor meta data are not supported in torch.compile. LieTensor uses additional attributes, e.g., the ltype, than a standard tensor, which breaks the compiler. It's hard to say whether this could be fixed because it depends on the torch dynamo.

lahavlipson, 2025-01-16:
I see. So it sounds like more of an issue with torch.compile than with pypose. Although it's strange that torch.compile throws an error rather than creating a graph break.

zitongzhan, 2025-01-16:
> I see. So it sounds like more of an issue with torch.compile than with pypose. Although it's strange that torch.compile throws an error rather than creating a graph break.

Right, it's supposed to fall back to the Python interpreter according to its design philosophy, but it throws the same error while we tested the code.

closed 2025-01-20 by lahavlipson